pikvm/pikvm #3: PCB Board Design to use in Lieu of breadboard

Makr91, 2019-12-15:
Hello, 

I've made the following PCB in EASYeda and order a few of the boards from JLCPCB. I am also waiting on the MOSFETS. Once those arrive I'll be able to confirm the linked PCB is a valid design. If you notice anything that should be changes please feel free to let me know.

https://easyeda.com/mark.gilbert/zerow-kvm-v1

mdevaev, 2019-12-15:
Neat! Thanks for your work. I recommend the use of a wider tracks, 20mil for example. This will reduce their resistance and the possibility of accidental damage during assembly.

mdevaev, 2020-05-04:
I added a link to your design in the documentation. Thank you again!

closed 2020-05-04 by mdevaev

wnark, 2021-03-08:
Hello, does your printed circuit board work properly?

wnark, 2021-03-08:
> 你好，
> 
> 我已经在EASYeda中制作了以下PCB，并从JLCPCB订购了一些板。我也在等待MOSFET。一旦这些到达，我将能够确认链接的PCB是有效的设计。如果您发现任何需要更改的内容，请随时告诉我。
> 
> https://easyeda.com/mark.gilbert/zerow-kvm-v1

Hello, does your printed circuit board work properly?

th3drk0ne, 2021-03-08:
i have a few pcb designs here and all circuits tested and working [images of the built boards can be seen on discord]

https://oshwlab.com/dunne.karl

ATX hat is here https://oshwlab.com/dunne.karl/pi-kvm-atx-hat-v1-0
alternative mosfet is the G3VM-61G1

wnark, 2021-03-08:
> i have a few pcb designs here and all circuits tested and working [images of the built boards can be seen on discord]
> 
> https://oshwlab.com/dunne.karl
> 
> ATX hat is here https://oshwlab.com/dunne.karl/pi-kvm-atx-hat-v1-0
> alternative mosfet is the G3VM-61G1

wow,thanks,i would try it 。thanks

wnark, 2021-04-07:
> i have a few pcb designs here and all circuits tested and working [images of the built boards can be seen on discord]
> 
> https://oshwlab.com/dunne.karl
> 
> ATX hat is here https://oshwlab.com/dunne.karl/pi-kvm-atx-hat-v1-0
> an alternative optocoupler is the G3VM-61G1

Hello, I printed the circuit board. In actual work, I can turn on, shut down, and restart the computer normally, but the two led lights on this circuit board will not light up, and I measured the voltage across the led lights and found that the voltage is 0 , What is the function of these two led lights?
[image]
[image]

th3drk0ne, 2021-04-07:
The LED pins sense the voltage from the motherboard they don't light up LEDs attached to the board. you piggyback the LED header pins from the motherboard. The status of the led is displayed on the UI (power symbol Green when power is on HDD symbol flashes red to display activity)
[image]

wnark, 2021-04-07:
> The LED pins sense the voltage from the motherboard they don't light up LEDs attached to the board. you piggyback the LED header pins from the motherboard. The status of the led is displayed on the UI (power symbol Green when power is on HDD symbol flashes red to display activity)
> [image]

I got it, thank you. No wonder there is a hard drive light sign on it

wnark, 2021-04-09:
> i have a few pcb designs here and all circuits tested and working [images of the built boards can be seen on discord]
> 
> https://oshwlab.com/dunne.karl
> 
> ATX hat is here https://oshwlab.com/dunne.karl/pi-kvm-atx-hat-v1-0
> an alternative optocoupler is the G3VM-61G1

Hello, can 2N7002H-7 in your Pi-KVM Pi4 Hat v1.1 be replaced with 2N7002 12w
The 2N7002H-7 in my place is too expensive

th3drk0ne, 2021-04-09:
that will work you could also use a BSS138

wnark, 2021-04-09:
> BSS138

ok thanks

th3drk0ne, 2021-04-09:
> > BSS138
> 
> ok thanks

if you are using it for the fan controller circuit I
 have a script that manages the fan speed https://github.com/th3drk0ne/DKN-Fan-Controller, adjust the pwm and temperatures in the python script to adjust the speed based on a curve

[image]

wnark, 2021-04-29:
> > > BSS138
> > 
> > 
> > ok thanks
> 
> if you are using it for the fan controller circuit I
> have a script that manages the fan speed https://github.com/th3drk0ne/DKN-Fan-Controller, adjust the pwm and temperatures in the python script to adjust the speed based on a curve
> 
> [image]

Hello, I still need to trouble you
May I ask this rst, when writing a program in the arduino, does it need to be connected or disconnected?
[image]
When simulating as a simulated device, do I need to disconnect or connect?

And when the computer writes programs to the arduino, do I need to disconnect the Raspberry Pi?
Thank you

th3drk0ne, 2021-04-29:
Disconnected while programming and connected during normal use I used a
jumper for this

th3drk0ne, 2021-04-29:
Also the pi programs the board so it needs to be connected via usb to the
pi then follow the programming steps