pikvm/pikvm #4: ASUS bios resolution glitch

mallendeo, 2020-02-02:
Hi!

There is a bug/glitch with ASUS bios when using the TC358743 chip (I'm using the Lusya bridge). My guess is because the "low" resolution (720p).

[image]

Is there any way to force the hdmi stream at 1080p at lower framerates?

Thanks!

mdevaev, 2020-02-02:
One of the users told me about a similar problem, but I don't know how to solve it yet. I think I had a motherboard somewhere with a similar BIOS. I'll see if something can be done in this case.

PS: Does this BIOS have the option to enable the legacy interface without graphcs?

mallendeo, 2020-02-02:
The bios doesn't have that option afaik. I also searched for an older bios version to flash but it seems they all comes with the UEFI menu.

PS: This is what I tried without luck

`/etc/kvmd/override.yaml`

```yaml
kvmd:
    streamer:
        cap_pin: -1
        cmd:
            - /usr/bin/ustreamer
            - --device=/dev/kvmd-video
            - --persistent
            - --format=uyvy
            - --encoder=omx
            - --glitched-resolutions=720x480,720x576
            - --workers=4
            - --quality={quality}
            - --desired-fps={desired_fps}
            - --resolution=1920x1080
            - --drop-same-frames=30
            - --last-as-blank=0
            - --unix={unix}
            - --unix-rm
            - --unix-mode=0660
            - --exit-on-parent-death
            - --process-name-prefix={process_name_prefix}
            - --no-log-colors
        desired_fps: 15
        max_fps: 15
        quality: 50
        state_poll: 1.0
        timeout: 2.0
```

mdevaev, 2020-02-03:
I found this: http://forum.asrock.com/forum_posts.asp?TID=3108&title=wrong-resolution-in-uefi-setup-menu

> I got the same problem with an Asrock Z170 Extreme 4 and a MSI 1060 Gaming X 3G, there was no high resolution in the BIOS, even if I selected "Auto" in the settings of UEFI. My display is an iiyama x2783hsu-dp which is connected via DP. I have read that te DP can cause this kind of failure, but I like the DP, and I didn't wanna change it to an other because of some seconds that isn't perfect in the boot sequence.

> Firstly I used a MSI Gaming X RX470 8G, and everything was fine with the resolution in the BIOS. Later I changed it to the 1060, and the resolution went to low. Each and every option was as the same, as the software and BIOS version too. I started with a BIOS upgrade, doesn't helped. I came here, but there was no solution. The I gave up and I left everything in that state, bytheway there was no any problem with my hw config except this. Till now...

> **I found a forum, where somebody said that try to disable the CSM (Compatility Support Module). If you have modern hw and os, that all support UEFI, than you can easily disable this option**. Officially the UEFI is supported _after_ win7, so you might hold it enabled when you are still using win7.

> Finally I turned the csm on off, and the wonder happend, both the bios and the boot screen are full hd :) So give it a try!

Please try it

mallendeo, 2020-02-03:
It works!

It's really weird though. If I plug a DP cable, change it to CSM and leave it enabled, reboot, then plug the hdmi (Lusya), it works flawlessly. However if I disable CSM and don't remove the DP cable, it works but a lower resolution. Removing the DP makes the bug appear again.

I'll look into that, but I'm ok leaving the DP connected, or using the CSM mixed mode for now.

Thank you so much @mdevaev!

closed 2020-02-03 by mallendeo

mdevaev, 2020-02-03:
Could you help me with this? I need to briefly compile your solution as a tip to add it to the documentation. English is not my native language, it is difficult for me to do this.

mallendeo, 2020-02-03:
Yeah, no problem! I will add it to the documentation and send the PR.

zsmbrvr, 2020-09-21:
@mallendeo Could you try my fix? https://github.com/pikvm/pikvm/issues/78
All I did is change the EDID so the default resolution is 1280x1024 and not 1280x720.