pikvm/pikvm #7: Hardware KVM doesn't recognise keyboard hotkey switching from pikvm

n1nj4888, 2020-02-16:
Hi There,

As discussed in Discord, I have the following setup with a pikvm (Pi4, B101) with HDMI and USB connected to a 4 port Hardware HDMI KVM switch. The Hardware KVM then connects to 2 physical hosts.

```
DELL Keyboard (physical, USB)---------------|
                                            |
                                            V   	       ---> Host 1 (HDMI and USB)
Pivkm (Pi4,B101, HDMI + USBC) ---> Hardware HDMI KVM Switch ---|
							       ---> Host 2 (HDMI and USB)
```

When I manually switch the input host (either on the Hardware KVM itself or with a physical Dell Keyboard plugged into the Hardware KVM) between Host 1 and Host 2, this works fine and pikvm streams the HDMI output from Host 1 / 2 as expected. The pikvm keyboard also sends the usual keyboard signals to the selected host.

When using the Dell physical USB keyboard attached to the hardware KVM, I can use the following "hotkey sequences" and the Hardware KVM responds as expected:

CTRL (wait < 5 secs), CTRL - Cycles between connected Hosts 1 and 2
CTRL+SHIFT+1 - Changes to Input Host 1
CTRL+SHIFT+2 - Changes to Input Host 2

The issue is that, while I can send normal keystrokes fine with pikvm/virtual keyboard, the above "hotkey sequences" are not picked up by the Hardware KVM when sent via pikvm?

As requested on Discord, here's the output of the usbhid-dump -s BUS:DEVICE command for the Dell Keyboard when:

(A) Run on Host 2 with the Dell Keyboard connected directly to Host 2
(B) Run on Host 2 with the Dell Keyboard connected via the Hardware KVM Switch

Thanks for taking the time to look into this!

[(A) usbhid-dump-host2-direct-connect-001-005.txt](https://github.com/pikvm/docs/files/4209335/A.usbhid-dump-host2-direct-connect-001-005.txt)
[(B) usbhid-dump-host2-via_KVM_switch-001-007.txt](https://github.com/pikvm/docs/files/4209336/B.usbhid-dump-host2-via_KVM_switch-001-007.txt)

mdevaev, 2020-08-20:
OK, it was found out with other users that this problem is related to the fact that KVM switches are too stupid to understand composite devices. On the Pi-KVM side, we found a solution in the form of using the hardware [Arduino HID](https://github.com/pikvm/pikvm/blob/master/pages/arduino_hid.md). Thus, the Pi-KVM is connected to the switch with two USB cables: one for the HID, the other for the virtual drive. On switches usually one of the USB ports is marked as dedicated for connecting the keyboard and mouse, and the HID needs to be connected there.

closed 2020-08-20 by mdevaev

jcormier, 2023-06-13:
Note I didn't notice at first but there is a guide on how to disable the mass storage so the USB doesn't show up as a composite device. https://docs.pikvm.org/msd/#disable-msd

> To disable mass storage emulation altogether, you can place the following piece of configuration into /etc/kvmd/override.yaml:
```
kvmd:
    msd:
        type:  disabled
```