pikvm/pikvm #8: Please help with an error during installation

bslava, 2020-02-17:
Hi,
I'm trying to compile OS on Windows subsystem for Linux and receiving the following error during installation:

**~$ sudo sh get-docker.sh**
\# Executing docker install script, commit: f45d7c11389849ff46a6b4d94e0dd1ffebca32c1
\+ sh -c apt-get update -qq >/dev/null
\+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
\+ sh -c curl -fsSL "https://download.docker.com/linux/ubuntu/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
curl: (7) Failed to connect to download.docker.com port 443: Connection refused
gpg: no valid OpenPGP data found.

What should I do to fix this issue?

Thanks!
Slava

mdevaev, 2020-02-17:
Hello. Try this:
```
wget https://yum.dockerproject.org/gpg
sudo apt-key add gpg
```

bslava, 2020-02-18:
The issue was related to missing proxy. You could close the ticket. Thanks for help!

bslava, 2020-02-18:
In addition, it would be helpful to change the installation procedure:
`sudo usermod -aG docker user`
to:
`sudo usermod -aG docker $USER`

It would save some time for inexperienced users, who just copy commands from your list one by one.

Thanks,
Slava

mdevaev, 2020-02-18:
Good point. Thanks!

closed 2020-02-18 by mdevaev