pikvm/pikvm #10: "Remote key not fetched correctly from keyserver" error during compilation.

bslava, 2020-03-13:
Hi,

I'm trying to compile the image and received the following error:
`---> Running in e505b452212b
gpg: keyserver receive failed: No keyserver available
==> ERROR: Remote key not fetched correctly from keyserver.
The command '/bin/sh -c pacman-key --keyserver pgp.mit.edu -r 912C773ABBD1B584 && pacman-key --lsign-key 912C773ABBD1B584 && echo -e "\n[pikvm]" >> /etc/pacman.conf && case "$BOARD" in zero|zerow) echo "Server = https://pikvm.org/repos/rpi" >> /etc/pacman.conf;; *) echo "Server = https://pikvm.org/repos/$BOARD" >> /etc/pacman.conf;; esac && echo "SigLevel = Required DatabaseOptional" >> /etc/pacman.conf' returned a non-zero code: 1
make[1]: *** [Makefile:208: os] Error 1`
What I'm doing wrong? Please help!

Regards,
Slava

mdevaev, 2020-03-13:
Hello! Just try build again, it looks like a temporary problem

bslava, 2020-03-13:
I found a w/a -t define BOARD variable from the command line. After that make completed. I do not understand why it wasn't initialized from config.mk file.

mdevaev, 2020-03-16:
Strange issue, no one has ever encountered this

JsBergbau, 2020-03-27:
I've encountered this issue several times when building the OS. After a second `make os` it went on and finally completed the build process.

mdevaev, 2020-03-27:
Still can't reproduce :/

dogmachine, 2020-04-23:
i ran into this.  retried 10 times and same problem. changed pgp.mit.edu to hkps://pgp.mit.edu and it worked.  but also the http frontend for http://pgp.mit.edu was sometimes failing to read from its upstream proxy so i suspect it was just semi-broken at the time.  probably would have worked without making that change had i tried again at that time.  maybe chuck some alternative key servers in there?

mdevaev, 2020-04-23:
Well, if hkps://pgp.mit.edu is working fine I will add it. Ironically, many people had a correlation of this problem with specifying the BOARD variable. I was beginning to think I was going mad.

mdevaev, 2020-04-23:
Wait, I understand correctly that you think that sometimes hkps does not respond?

mdevaev, 2020-04-23:
Yes, it is. Now we will use. hkps://hkps.pool.sks-keyservers.net
Thank you for the diagnostics!

closed 2020-04-23 by mdevaev