pikvm/pikvm #9: Instructions to connect B101 to RPI 4

bslava, 2020-03-13:
Hello,

Is there any instructions how to connect B101 to RPI 4? Will it work with the cable supplied with B101?

By the way, following the build instructions on Ubuntu, I received the following message:
`docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock:`

Since I'm not familiar with docker, it took me a while to find the solution:
`sudo chmod 666 /var/run/docker.sock`
I think this info could be helpful to other people.

Regards,
Slava

mdevaev, 2020-03-13:
Hi!

> Is there any instructions how to connect B101 to RPI 4? Will it work with the cable supplied with B101?

Yes, I checked

> sudo chmod 666 /var/run/docker.sock

This command from the manual should solve your problem: `sudo usermod -aG docker $USER`. But it seems that after this you need to perform a relogin. Could you restore your permissions and restart the machine? Everything should work fine.

bslava, 2020-03-13:
Thanks for so speedy response!

> Yes, I checked

Where is it? I do not see anything related to B101.

> But it seems that after this you need to perform a relogin.

I'd mention this relogin in the documentation.

Thanks!
Slava

mdevaev, 2020-03-13:
> I do not see anything related to B101.

Just connect using the wire and miracle will happen. The B101 is almost identical to the Chinese video capture board.

> I'd mention this relogin in the documentation.

Good point.

closed 2020-03-13 by bslava

mdevaev, 2020-03-13:
> I'd mention this relogin in the documentation.

Added.