pikvm/pikvm #11: Why splitting the cable for RPI4?

JsBergbau, 2020-03-26:
> RPi4 only: since it uses one USB-C female connector to giving power and keyboard/mouse/drive emulation you also need to make a special cable to split DATA and POWER lines from USB-C. You can make it from two suitable connecting cables, or solder from scratch. Be sure to check the circuit diagram, otherwise you may damage your devices. Pinout specific used connectors you can easily find on request "USB pinout" in Google. Please note that if you will make a cable based on the two factory ones, the colors of the wires may not match those shown in the picture. Use a multimeter to make sure the connections are correct.

Why do you recommend splitting the cable? On a lot of PCs you can set that USB ports get always power even in OFF state. You can set that in BIOS. So I just want to understand the background of your advice.

With a good quality USB-C cable connected to a USB3.0 Port it provides even enough power that the PI doesn't print its undervolted message.
If power shouldn't be sufficient I would recomment using backpowering USB-A to USB-A cable. It is not possible to power it this way, but only to give it additional power. Even my RPI4 gets enough power via the PC USB3.0 connection it still draws 270 mA (=1,35W) when using backpower. 

It works so smooth, just install, connect and you're ready. Splitting the cable is a effort you can easily avoid that way.

mdevaev, 2020-03-27:
The reason is that not all motherboards provide power in the off state. In addition, the USB current may not be sufficient for both stable RPI power supply and video capture operation. And in the latter case, you will not see any warnings from the kernel - just NO SIGNAL on the screen, this is a feature of the capture device.

If you want to bring power to the back panel via USB-A, then you may expect another surprise: some motherboards refuse to boot if the power to the USB is supplied from outside. For the same reason, you can't power the RPI via GPIO pins and just connect the USB to the computer - the current will flow towards it and you will get a problem when booting.

In short, our instructions describe how to make a device that is guaranteed to work in all cases. If the user wants to change something in the design, they must clearly understand what they are doing.

JsBergbau, 2020-03-27:
> If you want to bring power to the back panel via USB-A, then you may expect another surprise: some motherboards refuse to boot if the power to the USB is supplied from outside. For the same reason, [...]

Thanks. I see you're a real expert. Didn't know that some mainboards won't boot then.

mdevaev, 2020-03-27:
Oh, come on. I'm just lucky to have strange problems :)

closed 2020-03-27 by mdevaev

twisteroidambassador, 2021-01-27:
I think this may technically be a USB protocol violation. A self-powered USB device is not supposed to output on D+/D- unless a host is present, and on USB 2.0 the device can only detect the host by the 5V line being high. When the Pi is powered via the split cable, 5V is always high, therefore it may output on D+/D- even when the server is not turned on / not powering its USB port.

The above is assuming Pi 4 does not do the proper Type C host/device detection, and apparently it is the case: https://www.raspberrypi.org/forums/viewtopic.php?t=246348#p1504436

I'm sure the current splitter design works well in practice. Still, I wonder whether there's a more compliant solution.

mdevaev, 2021-01-27:
I know what you mean but there is no other solution at the moment as the pi power supply circuit is inseparable from the 5v detection. The USB controller is smart enough to handle this situation and it's better than backpowering.

outrullers, 2021-01-28:
> For the same reason, you can't power the RPI via GPIO pins and just connect the USB to the computer - the current will flow towards it and you will get a problem when booting.

That is the exact way I want to provide power to my Pi-KVM.
My setup:
1. power RPi-4 thru GPIO using standalone 5V PSU
2. connect RPi-4 USB port to server`s USB port

Can I add a diode to prevent current flow from RPi-4 USB towards the server`s USB port?

wzqvip, 2023-11-23:
That's fine. It's kinda optional. For some motherboard like "JINGYUE", if external power supply connected, the current will power the cooling fan, and for some reason it will prevent bootup. One lazy solution is not to connect the USB to the system first, use external 5V. After poweron, connect to PC and (may) remove the external power supply.