pikvm/pikvm #12: Options for powersave

JsBergbau, 2020-03-27:
FIrst I want to thank you for your great project. You provide exactly the thing I was looking for. Thank you very much for all your effort you put into this project.

Powering Off HDMI saves 25 mA on the PI Zero W. Because powersupplies don't have 100 % efficiency you save 0.15-0.2 Watt. It sounds not a lot, but in sum it is a lot.
99 % of users won't need the HDMI output of the KVM Raspberry PI.
So we could savely turn off HDMI via `/opt/vc/bin/tvservice -o`
On Raspian I do this with a crontab "@reboot" line, however I'm not firm with ArchLinux. So it would be great if you add an option to disable HDMI.

mdevaev, 2020-03-27:
Okay, that makes sense. This option must be in config.mk what do you think?

JsBergbau, 2020-03-27:
Yeah, config.mk is the right place.

mdevaev, 2020-04-27:
Sorry for the delay.

I tried using this and found out that disabling HDMI can cause the RPi to hang on boot. I believe that you have this working in a particular case, but I would not like to add an option that can cause such problems.

mdevaev, 2020-04-27:
UPD: I did this: https://github.com/pikvm/packages/blob/master/packages/raspberrypi-hdmi/raspberrypi-hdmi-off.service

JsBergbau, 2020-04-28:
Hi. I have 4 PIs running 24/7 (except updates) and never hang on boot. What I do is to poweroff HDMI about 120-300 seconds after boot. This gives enough time to disable this feature if anything goes wrong at boot. Then you still see the output and could disable this feature before screen goes off.

E.g. thats my solution for disabling output

`@reboot sleep 120s && /opt/vc/bin/tvservice -o`

mdevaev, 2020-04-28:
I understand correctly that you also got a hang-up during boot, so you added a delay?

JsBergbau, 2020-04-28:
I've never tested the variant with the service. I had never a hang on boot because of disabling HDMI. It's just a safety measure. If any error occurs on boot, I can connect a monitor and see whats going on. All my raspberry PIs are headless, but if I ever had to configure them without network, I can attack a monitor and keyboard, then boot, disabling the delayed HDMI power off and then doing the necessary maintenance work.

mdevaev, 2020-04-28:
You can try the service. I made a package for this. Install `raspberrypi-hdmi` using pacman.

To enable HDMI:
`systemctl start raspberrypi-hdmi-off`; to disable stop the service. To disable during boot: `systemctl enable raspberrypi-hdmi-off`. But be prepared to hang :)

mdevaev, 2020-06-28:
Okay, I've just finished package for this feature. You can install it and enable like this:
```
pacman -S raspberrypi-hdmi
systemctl enable raspberrypi-hdmi-off.timer
```
**Don't** enable raspberrypi-hdmi-off.service, it may cause a hang on boot.

Since this option is potentially dangerous and may cause end-user questions, I would prefer not to enable it by default and not add the parameter to the build environment. I assume that for non-standard pikvm configurations, you can edit Dockerfile.part yourself to install the desired package.

closed 2020-06-28 by mdevaev

JsBergbau, 2021-03-10:
Just testet. Works like a charm.

bunducafe, 2024-03-20:
Unfortunately the package cannot be installed with the latest version as it conflicts with praspberrypi-utils.

Is there a workaround? Or is just manually disabling doing the trick here?

mdevaev, 2024-03-20:
@bunducafe It made sense for Zero, which is no longer supported. Why do you need this now on modern boards? 0.1W is a drop in the ocean.

bunducafe, 2024-03-20:
Actually, I found that very handy when it comes to c-states on my server. The piKVM does prevent that. Once I unplug the HDMI cable c8 / c10 states are reached immediately.... But now I limit myself to just being able to ssh into the server and start / stop it. That's why I was asking.

The other remedy would be to just switch off the hdmi port completely.

mdevaev, 2024-03-20:
Sorry, I don't understand anything about states. Could you explaing in details?

mdevaev, 2024-03-20:
Oh, I get it. This package has nothing to do with HDMI communication with the server. This was necessary to turn off the HDMI **output** on the PiKVM itself.

bunducafe, 2024-03-20:
Yes, that's what I understood. I will then disable the tvservice on the pi temporarily.

mdevaev, 2024-03-20:
This is no longer supported after Zero is decommissioned.