pikvm/pikvm #13: Option for inverting GPIO relay logic

mallendeo, 2020-04-29:
Hi, the [optocoupler](https://www.aliexpress.com/item/32964487766.html) I bought isn't working for some reason, so I'm using this relay instead.

[image]

For the moment I am manually editing the `set_output` and `write` methods on
`/usr/lib/python3.8/site-packages/kvmd/gpio.py` for this to work. Otherwise it will initialized with the on state and my PC won't be able to start. Inverting the logic at the relay as "normally open" will work only if the rpi is on.

We could use an ENV variable, like `GPIO_INVERT_WRITE` and `GPIO_INVERT_READ`.

mdevaev, 2020-04-29:
> Inverting the logic at the relay as "normally open" will work only if the rpi is on.

I didn't understand the problem. You need to have the relay normally open by default to simulate pressing a button. I didn't understand the problem. You need to have the relay normally open by default to simulate pressing a button. Your relay has a normally open channel, why not use it?

mallendeo, 2020-04-29:
Sorry, my bad. It's actually normally open. When kvmd starts, the relay is closed without inverting the logic.

[image]

mdevaev, 2020-04-30:
What is the status of the relay before starting kvmd? Starting kvmd immediately changes the relay state from open to closed?

mallendeo, 2020-04-30:
Before starting kvmd it's off (open), and yes, it immediately changes to closed once the rpi has booted.
With the change above, it keeps its initial state after boot.

mdevaev, 2020-04-30:
Ah, I see. Your relay is controlled by zero level. This means that in any uncertain situation, it can close and unexpectedly turn off / turn on your server. I strongly do not recommend using this module. If you are determined to use this particular relay, I will add an option for you. But this is a big source of problems.

mallendeo, 2020-05-01:
I'm using it only because I didn't have any luck trying with an optocoupler or the 817 module (rpi gpio output is 3.3v and the 817 module requires 5v).

The relay has been working well for now, although thinking about it again, if this option does not makes any sense for the project or makes the code harder to reason about, I wouldn't worry (since this is a workaround for my use case). Thanks anyway for your time! :)

mdevaev, 2020-05-01:
Relay modules for 5V should work well. Give me a link to your module, and I will see what the problem is with my hardware engineer comrade. Also show which optocouplers you use.

mdevaev, 2020-05-01:
Adding this feature is not very difficult, but I would prefer to help you with your problem rather than treat its symptoms :)

mallendeo, 2020-05-01:
Thank you. I really appreciate this!

These are the ones I tried first
[image]

Then I bought this board
[image]

It lights up a little, but it doesn't have enough power to fully activate the sensor with 3.3v.

mdevaev, 2020-05-01:
Okay, here's the thing. Your optocouplers are based on transistors. The module you bought uses the same optocouplers that you tried initially. Polarity is also very important for them. Now you have these options:

1. You may buy optocouplers OMRON G3VM-61A1 as recommended by our instruction. This is not too expensive and we have selected these parts according to their characteristics so that they work well with RPi.
2. You can also try to modify the mechanical relay module so that it is controlled by a non-zero signal. This will require some soldering and will deprive the module of its factory beauty, but it should work.

In any case, I suggest you go to our discord, where we will help you faster: https://discord.gg/bpmXfz5

mallendeo, 2020-05-05:
Sorry for the late response.

> You may buy optocouplers OMRON G3VM-61A1 as recommended by our instruction.

Awesome!, because I ordered two of them a month ago, they seem stuck at customs though :p

Can I just replace the optocouplers with the ones in the 817 module and use the same circuit? (at least for the switches)

mdevaev, 2020-05-05:
No, I told you. They use the same optocouplers that you've tried before. Look at the markings on their body.

mallendeo, 2020-05-05:
No no, I mean these hahaha

[image]

mdevaev, 2020-05-05:
Sorry. Yes, it will work.

mdevaev, 2020-05-12:
I added information to the documentation that you need to use normal relays. I think this issue can be considered resolved.

closed 2020-05-12 by mdevaev

mallendeo, 2020-05-21:
Just wanted to say thank you so much! The optocouplers finally arrived and it's working now, I couldn't make it to work with the existing PCB so I put a 330 Ohm resistor and it worked flawlessly.

mdevaev, 2020-05-21:
Cool!

myself248, 2020-08-24:
Not sure where else to put this, but in every motherboard I've looked at, there's no need for optocouplers. Barely even a transistor, but transistors are pennies. The Power and Reset signals are floating, and their respective pushbuttons just ground them. Since the Raspberry Pi shares a ground with the PC being controlled, these can be tied directly to GPIO. I demonstrate it here:

https://www.youtube.com/watch?v=eXzK3U6huwY 

The gate charge of those FETs is only a few pF so a gate resistor isn't even needed. Doesn't get much simpler. (Arguably it does, if you had 5v-tolerant GPIO like certain knockoff boards...)

mdevaev, 2020-08-24:
@myself248 The relays are used for galvanic isolation in case the KVM and server are connected to different power outlets. And if you mix up the poles, you can ruin something on both the motherboard and the Pi. You can use this method if you have checked everything carefully, but remember that this way you lose additional protection from errors. And you can manage something non-standard, not quite ATX. In any case, not using the relay is a great way to see the magic smoke.

somuchtyler, 2020-10-22:
@mdevaev just to double check, i've been using optocouplers (PC817 are really common on aliexpress) for a project a while back and it seemed to work fine and what I do is wire them like the attached picture, just wanted to double check if there's any downsides to using this setup...? the reason I do it this way is so I don't need to check for polarity at all
[image]

mdevaev, 2020-10-23:
@somuchtyler this circuit should work fine, but you need to hang a separate resistor on each anode to limit the current. The value of the resistor must be calculated based on the characteristics of the optocoupler.

somuchtyler, 2020-10-23:
@mdevaev good catch, I did forgot to put in resistors before each of the anodes in my picture. If doing 3.3V logic with a typical forward voltage of 1.2V and a max current of 50mA. If wanting to stay under that at say at 20mA, it would be 105ohms right? 
(3.3-1.2)/0.020=105ohms

mdevaev, 2020-10-23:
@somuchtyler yep. 120 ohm is standard value. You can usse it I guess.

atesin, 2020-11-14:
hi ...  sorry i didnt read completely, but working with arduino i also had some troubles with these chinese relays, too bad they are poorly documented anyway

searching a lot i discover the *correct* way to use it ....  the mechanical module (the blue little box) is a coil relay with 12v sensitivity, so to get work with 5 or less volt were mede with optocopulers in the middle, so you have to feed the optocopulers with a little voltage (5v-2.5v?) and the coil relays with 9v-12v isolated

but the relays module comes by default with these 2 circuits shorted with a jumper, thats because 5v inputs are not enough to trigger the relays (or unpredictable)...  so you have to remove the jumper and feed with 12v apart to get the module work properly

```
     +---------+                            
     | 220v ac |--------------------------, 
     |         |------------------,        |
     +---------+                   | loads |
                     +----------+  |   |   |
                     |          |  |   v   |
      +--------+     |        k1|--+       |
      | 12v dc |-----|gnd       |--|--(@)--+
      |        |-----|jd-vcc    |  |       |
      +--------+     |        k2|--+       |
                    ,|vcc       |--|--(@)--+
                   / |gnd       |  |       |
+--------------+  / ,|in1     k3|--+       |
|              | / / |in2       |--|--(@)--+
|            5v|' /  |in3       |  |       |
| arduino   gnd| /   |in4     k4|-'        |
|          d1-4|'    |          |-----(@)-'
|              |     +----------+           
+--------------+
```

DocJr90, 2024-01-12:
@mdevaev It appears the G3VM-61A1 has been EOL'ed. Any suggestions on a specific replacement? I'm currently searching through Omron's products but the naming scheme isn't particularly obvious and/or helpful to a novice.

DocJr90, 2024-01-12:
@mdevaev Finally found it after manually navigating Omron's website and locating the datasheet instead of looking for discontinuation notices. Mouser also wasn't immediately helpful due to the generic label ("Solid State Relays - PCB Mount MOS-FET RELAY"). The recommended part number that appears equivalent for new designs is:  G3VM-61AY1

mdevaev, 2024-01-12:
Added to the list, thank you

HotelMoted, 2024-01-13:
I've been using 2x PC817 per power switch for years now and never had a problem across dozens servers.

jiaxinonly, 2024-09-20:
> Ah, I see. Your relay is controlled by zero level. This means that in any uncertain situation, it can close and unexpectedly turn off / turn on your server. I strongly do not recommend using this [...]

Sorry, I do need an option to inverted power_switch_pin and reset_switch_pin, I need to control atx directly using gpio without adding any other relays. I wanted to do it despite the risks I had seen many times in the issue. The user can be strongly warned in the document, but the user can choose whether to use it or not

mdevaev, 2024-09-20:
I don't want to add solutions that allow the user to shoot himself in the foot. In my experience, such options always raise too many support questions and related problems.

jiaxinonly, 2024-09-20:
Well, you're right. It does cause unnecessary trouble :joy:

jiaxinonly, 2024-09-20:
If you need to inverted power_switch_pin and reset_switch_pin, you can do so, but make sure you are aware of the risks before doing so. (System demonstration based on pikvm v4 plus)
:warning: It is strongly not recommended that you do this！:warning: 
1. ssh connects to pikvm
2. change the system to rw
    ```shell
    rw
    ```
3. edit /usr/lib/python3.12/site-packages/kvmd/plugins/atx/gpio.py
    ```shell
    vim /usr/lib/python3.12/site-packages/kvmd/plugins/atx/gpio.py
    ```
3. change the default level of power_switch_pin and reset_switch_pin to high
    ```python
    def sysprep(self) -> None:
        assert self.__line_req is None
        self.__line_req = gpiod.request_lines(
            self.__device_path,
            consumer="kvmd::atx",
            config={
                (self.__power_switch_pin, self.__reset_switch_pin): gpiod.LineSettings(
                    direction=gpiod.line.Direction.OUTPUT,
                    output_value=gpiod.line.Value(True),  # Set to high
                ),
            },
        )
    ```
4. modify the click logic. After clicking, the output is low and then restored to high
    ```python
    async def __inner_click(self, name: str, pin: int, delay: float) -> None:
        assert self.__line_req
        try:
            self.__line_req.set_value(pin, gpiod.line.Value(False))  # Set to low when you click
            await asyncio.sleep(delay)
        finally:
            self.__line_req.set_value(pin, gpiod.line.Value(True))  # Restore to high level
            await asyncio.sleep(1)
        get_logger(0).info("Clicked ATX button %r", name)
    ```
5. restore the system to read-only
    ```shell
    ro
    ```
6. restart the kvmd service
    ```shell
    systemctl restart kvmd
    ```

itiberiu, 2024-09-20:
ty very much! now my relay is premanent off, except when i use it!

itiberiu, 2025-04-19:
I just want to add a change. In the meantime, the path to the file has changed.

vim /usr/lib/python3.13/site-packages/kvmd/plugins/atx/gpio.py