pikvm/pikvm #14: Enable Lowcost S-Video support for V2 Platforms

MarkProminic, 2020-05-02:
In the Documentation you state that S-Video capture only works for the v2 Platforms. 

I would however like to use a Pi Zero W, with a USB Video Capture UTV 007. Can you clarify why it isn't possible to use these on the V2 Platform devices?

mdevaev, 2020-05-02:
The reason is that you will have to use a single USB port of Zero to connect the video capture device. Since it's highly likely that you will want to use the USB hub to connect something else, we cannot predict which port the video capture will be connected to. This means that we will not be able to write the correct configuration file for this.

In addition, you will have to use Arduino to emulate the keyboard, since OTG will not be available (remember, you took up the only USB port? It can only work in one mode: host or OTG, one excludes the other).

The price of a device with a cheap video capture, Arduino and other parts is approximately equal to the price with HDMI, while the second is much easier to manufacture and has more features. Therefore, we suggest users to do this either on the RPi3 v1 or v2 to Zero.

MarkProminic, 2020-05-04:
Well, I was actually hoping to find a way to use the Pi Zeros from the DIY-IPMI project, and using a VM on one of the other hosts that will do the actual video processing since the Pis have limited CPU and I can increase the amount of CPU as needed on a VM. 

It makes sense that you have to use the HDMI module when you are setting up a singular Pi0 KVM, however if we are installing a cluster of them, like in the DIY-IPMI project, I would be able to centrally manage them.

mdevaev, 2020-05-04:
Okay, I get it. Well, you should understand that your scenario is too special to support something like this out of the box. At least at this stage of this project's development. I suppose you can study the kvmd configuration files and try writing your own version for your case. You can start with the v2 build as the closest to your configuration, and then start changing the parameters: it will change hid plugins, video capture devices, and so on. It seems that technically all the necessary things are already supported and will not be removed in the future.
https://github.com/pikvm/kvmd/tree/master/configs

mdevaev, 2020-05-04:
You can also connect to our discord chat. There are many enthusiasts of this project who can help you, besides me. https://discord.gg/bpmXfz5

mdevaev, 2020-06-03:
No more questions, it seems :) Feel free to reopen this issue.

closed 2020-06-03 by mdevaev