pikvm/pikvm #15: Link libjson-c.so.4 to libjson-c.so.4.0.0 is missing

stsirakidis, 2020-05-05:
Fyi, there is a link missing and the kvmd-webterm service cannot be started:

```
[root@pikvm ~]# systemctl status kvmd-webterm
* kvmd-webterm.service - Pi-KVM - Web terminal (ttyd)
     Loaded: loaded (/usr/lib/systemd/system/kvmd-webterm.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2020-04-25 01:53:57 CEST; 47s ago
    Process: 327 ExecStart=/usr/bin/ttyd --interface=/run/kvmd/ttyd.sock --port=0 /bin/bash -c echo -ne "0;Pi-KVM Terminal: `hostname -f` (ttyd)^G"; cat /etc/motd; bash (co>
    Process: 328 ExecStartPost=/bin/bash -c sleep 3; chmod 660 /run/kvmd/ttyd.sock (code=exited, status=1/FAILURE)
   Main PID: 327 (code=exited, status=127)

Apr 25 01:53:54 pikvm.domain systemd[1]: Starting Pi-KVM - Web terminal (ttyd)...
Apr 25 01:53:54 pikvm.domain ttyd[327]: /usr/bin/ttyd: error while loading shared libraries: libjson-c.so.4: cannot open shared object file: No such file o>
Apr 25 01:53:54 pikvm.domain systemd[1]: kvmd-webterm.service: Main process exited, code=exited, status=127/n/a
Apr 25 01:53:57 pikvm.domain systemd[1]: kvmd-webterm.service: Control process exited, code=exited, status=1/FAILURE
Apr 25 01:53:57 pikvm.domain systemd[1]: kvmd-webterm.service: Failed with result 'exit-code'.
Apr 25 01:53:57 pikvm.domain systemd[1]: Failed to start Pi-KVM - Web terminal (ttyd).
```
Adding the link will solve the problem:
```
[root@pikvm lib]# rw
[root@pikvm lib]# ls -al  libjson*
lrwxrwxrwx 1 root root    18 May  5  2020 libjson-c.so -> libjson-c.so.4.0.0
-rwxr-xr-x 1 root root 54780 Apr 30  2020 libjson-c.so.4.0.0
lrwxrwxrwx 1 root root    18 May  5  2020 libjson-c.so.5 -> libjson-c.so.5.0.0
-rwxr-xr-x 1 root root 54780 Apr 30  2020 libjson-c.so.5.0.0
[root@pikvm lib]# ln -s /usr/lib/libjson-c.so.4.0.0 /usr/lib/libjson-c.so.4
[root@pikvm lib]# ls -al  libjson*
lrwxrwxrwx 1 root root    18 May  5  2020 libjson-c.so -> libjson-c.so.4.0.0
lrwxrwxrwx 1 root root    27 Apr 25 01:57 libjson-c.so.4 -> /usr/lib/libjson-c.so.4.0.0
-rwxr-xr-x 1 root root 54780 Apr 30  2020 libjson-c.so.4.0.0
lrwxrwxrwx 1 root root    18 May  5  2020 libjson-c.so.5 -> libjson-c.so.5.0.0
-rwxr-xr-x 1 root root 54780 Apr 30  2020 libjson-c.so.5.0.0
[root@pikvm lib]# systemctl start kvmd-webterm
[root@pikvm lib]# ro
```

mdevaev, 2020-05-05:
Thanks! Fixed. Remove the link and just upgrade the system using `pacman -Syu`.

closed 2020-05-05 by mdevaev

stsirakidis, 2020-05-05:
Thanks for the quick fix and for this great project! 👍