pikvm/pikvm #16: /dev/mmcblk0p3 on /var/lib/kvmd/msd turns to read-only during upload of an image

stsirakidis, 2020-05-05:
The /var/lib/kvmd/msd turns to read-only during upload of an image and the upload breaks.

Here the dmesg:

[   60.115347] EXT4-fs (mmcblk0p3): re-mounted. Opts: errors=remount-ro,data=journal

mdevaev, 2020-05-05:
I need more diagnostics. What size is the image? When does the download stop? What is in the kvmd logs? `journalctl-u kvmd`

stsirakidis, 2020-05-05:
The size of the image seems to be not relevant...I tried with 300MB up to 700MB iso's....same case

````
May 05 19:42:23 pikvm.domain kvmd[400]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.316      http] -- HTTP: Disconnected the stream cl>
May 05 19:42:23 pikvm.domain kvmd[400]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- refe>
May 05 19:42:23 pikvm.domain kvmd[400]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.325      http] -- HTTP: Registered the new stream >
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RW ...
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-h>
May 05 19:42:24 pikvm.domain sudo[462]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount rw
May 05 19:42:24 pikvm.domain sudo[462]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 05 19:42:24 pikvm.domain sudo[462]: pam_unix(sudo:session): session closed for user root
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RW-mode ...
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/images
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/meta
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RO ...
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-h>
May 05 19:42:24 pikvm.domain sudo[473]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount ro
May 05 19:42:24 pikvm.domain sudo[473]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 05 19:42:24 pikvm.domain kvmd[400]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.329      http] -- HTTP: Disconnected the stream cl>
May 05 19:42:25 pikvm.domain kvmd[400]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- refe>
May 05 19:42:25 pikvm.domain kvmd[400]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.347      http] -- HTTP: Registered the new stream >
May 05 19:42:25 pikvm.domain sudo[473]: pam_unix(sudo:session): session closed for user root
May 05 19:42:25 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RO-mode ...
May 05 19:42:25 pikvm.domain kvmd[400]: aiohttp.access                    INFO --- [admin (token) / 10.55.160.12] 'POST /msd/remove?image=FreeNAS-11.3-U2.1>
May 05 19:42:33 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RW ...
May 05 19:42:33 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-h>
May 05 19:42:34 pikvm.domain sudo[484]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount rw
May 05 19:42:34 pikvm.domain sudo[484]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 05 19:42:34 pikvm.domain sudo[484]: pam_unix(sudo:session): session closed for user root
May 05 19:42:34 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RW-mode ...
May 05 19:42:34 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/images
May 05 19:42:34 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/meta
May 05 19:42:34 pikvm.domain kvmd[400]: kvmd.apps.kvmd.api.msd            INFO --- Writing image 'FreeNAS-11.3-U2.1 (1).iso' to MSD ...
May 05 19:42:36 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg              INFO --- Closing new image file ...
May 05 19:42:36 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RO ...
May 05 19:42:36 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-h>
May 05 19:42:36 pikvm.domain sudo[496]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount ro
May 05 19:42:36 pikvm.domain sudo[496]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 05 19:42:36 pikvm.domain sudo[496]: pam_unix(sudo:session): session closed for user root
May 05 19:42:36 pikvm.domain kvmd[400]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RO-mode ...
May 05 19:42:36 pikvm.domain kvmd[400]: kvmd.apps.kvmd.api.msd            INFO --- Written image 'FreeNAS-11.3-U2.1 (1).iso' with size=589824 bytes to MSD
````

mdevaev, 2020-05-05:
I understand correctly that it was interrupted somewhere on 70% of the download? Does the interface say that the image is corrupted?

stsirakidis, 2020-05-05:
It says actually that is broken...I can see the image in the drop down menu from "Image"...stating "640KiB, broken"

mdevaev, 2020-05-05:
Could you turn on the developer tools in your browser and see what happens to download requests in the network tab?

mdevaev, 2020-05-05:
What browser do you use? Interested in including the version.

stsirakidis, 2020-05-05:
I tested Firefox & Chrome. Same situation in both cases. From the developer tools (sorry I am not an expert), I could see in the network tab that 356MB (around half of the image) have been transferred:
````
Host: pikvm.domain
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:72.0) Gecko/20100101 Firefox/72.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
X-Requested-With: XMLHttpRequest
Content-Type: multipart/form-data; boundary=---------------------------191691572411478
Content-Length: 785164630
Origin: https://pikvm.domain
Connection: keep-alive
Referer: https://pikvm.domain/kvm/
Cookie: auth_token=bea34f089ddb723e4bc6cd3648ea66673d3392abaecd83efc92bf8ec1ea0955f
TE: Trailers
````

stsirakidis, 2020-05-05:
Now 10.25MB of the ISO image is shown (also as broken) in the "Image". Could it be that the nginx is missing a parameter for the max size of the upload image?

mdevaev, 2020-05-05:
Is there any way I can access your device over HTTP to perform diagnostics?

stsirakidis, 2020-05-05:
Sorry, but it is not possible, since it is within my corporate network.

mdevaev, 2020-05-05:
Ok, look. I am interested in the request and response to the "write" handle. Please send it to me so that I can understand what happens when the upload is interrupted.

[image]

stsirakidis, 2020-05-05:
I obfuscated the real hostname:

[image]

mdevaev, 2020-05-05:
This is looks like a network problem - the connection is lost and the download is interrupted. I was especially concerned about the large number of reconnections to the video stream - this is one of the indicators of network problems.

What RPi do you use? Zero, RPi4 or the other? Try reducing the percentage of stream quality to a minimum and try downloading again.

I know that there may be certain problems with uploading images larger than 4 GB (browser features, the reasons are being investigated), but small images should upload without problems. I just checked it on my device.

stsirakidis, 2020-05-05:
I use a RPi 4B. When I try to reduce the stream quality, it doesn't work. From dmesg:

````
[  178.270874] 1f40: 00000000 00088cb8 eff550f4 c021a420 00000000 c1400000 c1404fb0 c1404ff8
[  178.288902] 1f60: c14d9580 c126ce48 30c5387d 00000000 c1500e10 c1401f90 c0209268 c020926c
[  178.306991] 1f80: 60070013 ffffffff
[  178.320471] [<c0201a78>] (__irq_svc) from [<c020926c>] (arch_cpu_idle+0x38/0x3c)
[  178.337882] [<c020926c>] (arch_cpu_idle) from [<c025ef38>] (do_idle+0xc4/0x148)
[  178.355092] [<c025ef38>] (do_idle) from [<c025f2a8>] (cpu_startup_entry+0x18/0x20)
[  178.372530] [<c025f2a8>] (cpu_startup_entry) from [<c1200f90>] (start_kernel+0x500/0x540)
[  178.480634] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  178.495735] tc358743 10-000f: i2c_wr: writing register 0x500 from 0xf failed
[  179.520647] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  179.535793] tc358743 10-000f: i2c_wr: writing register 0x500 from 0xf failed
[  180.560663] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  180.575765] tc358743 10-000f: i2c_wr: writing register 0x500 from 0xf failed
[  181.600674] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  181.615807] tc358743 10-000f: i2c_wr: writing register 0x500 from 0xf failed
[  181.633387] tc358743 10-000f: i2c_rd: reading register 0x14 from 0xf failed
````

Btw, I have a B101 Revision 4. I think that the RPi or the B101 could have a problem. What do you think?

mdevaev, 2020-05-05:
It's a new kernel. Reproduced; please wait for fix.

stsirakidis, 2020-05-05:
I am glad to hear that you can reproduce it 👍

mdevaev, 2020-05-05:
Ok, upgrade OS using `pacman -Syu` and kernel panic will gone.

The new 5.x kernel is broken at several points at once (i2c, initrd build, etc.), so for now I have forced the use of the 4.x. After some time, the 5.x core will be fixed and I will remove this restriction. RPi forums say that the new release of the kernel is experimental. So let's live on the old kernel for now.

stsirakidis, 2020-05-06:
I did the upgrade and there were still some problems...so I recreated a new SD card and fyi there were some errors:

````
==> Creating gzip-compressed initcpio image: /boot/initramfs-linux.img
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
````
and
````
Step 77/77 : RUN sed -i -e "s/^#!!!CheckSpace/CheckSpace/g" /etc/pacman.conf
 ---> Running in 06fdc7d87d25
ERROR: ld.so: object '/usr/local/lib/libpreload-semop.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/local/lib/libpreload-semop.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
````

As for now:
-  the streaming is stable
- regarding reducing the streaming quality, I noticed the following issue: the streaming default quality is set to 80%. I can reduce only once to a lower value or increase to a higher value. Change the quality (after the first change) afterwards is not possible....after rebooting it is possible to change the value again...but only once again.
- the problem with uploading still exists, even if I reduce the quality to 5%
- by updating:
````
[root@pikvm ~]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: installing linux-raspberrypi4 (5.4.35-1) breaks dependency 'linux-raspberrypi4=4.19.118-1' required by kvmd-platform-v2-hdmi-rpi4
:: installing linux-raspberrypi4-headers (5.4.35-1) breaks dependency 'linux-raspberrypi4-headers=4.19.118-1' required by kvmd-platform-v2-hdmi-rpi4
````

mdevaev, 2020-05-06:
> ==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly

Upstream kernel package a little broken. They knows.

> ERROR: ld.so: object '/usr/local/lib/libpreload-semop.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.

Just ignore :)

>  Change the quality (after the first change) afterwards is not possible

Explain in more detail what happens when this happens?

> by updating

This is a temporary measure to prevent updating to a newer core that has a broken video capture device.

stsirakidis, 2020-05-06:
> Explain in more detail what happens when this happens?

I can adjust the quality from 80% e.g. to 5%. If I move the pointer to a different value, then the stream stops (seems to try to render to the new quality value), but the pointer turns back to the previous value e.g. again back to 5%.

mdevaev, 2020-05-06:
This is strange. No one has ever encountered something like this. But it does look like a problem with the network, as strange as it sounds. Could you plug an ethernet from your computer directly into the RPi to rule out my hypothesis?

mdevaev, 2020-05-06:
By the way, what happens if you change the broadcast quality and then reload the interface page? Will the quality change work again?

stsirakidis, 2020-05-07:
> This is strange. No one has ever encountered something like this. But it does look like a problem with the network, as strange as it sounds. Could you plug an ethernet from your computer directly into the RPi to rule out my hypothesis?

Sure...I did it, but it failed unfortunately to upload the image, too. Here was the "speed" more realistic regarding the upload speed (when you see how quickly the image is uploaded on the website). It could upload around 160MB from 700MB....

mdevaev, 2020-05-07:
Now you have me even more puzzled.

Okay, I realized that there is no access to the corporate network. But could you take the device home to give me access to it? I want to try to reproduce this myself, but there is no such problem on my devices.

stsirakidis, 2020-05-07:
My last comment regarding the "emulated speed of the uploaded data" was the solution.... I remounted /var/lib/kvmd/msd in RW and upload an ISO with SCP and that worked fine...without any errors in the switches traffic and so on.

Let me please say first....that you cannot imagine how many tests I have made so far...with switches (traffic monitoring/packet error capturing), network cable checks and so on... :-)

Anyway, the problem was the Kaspersky Antivirus. I have been testing the whole time with Firefox & Chrome (behind a proxy and without a proxy)...but the Kaspersky was "injected" in the traffic in between. After stopping Kaspersky, everything worked perfectly! Thank you for your assistance so far!

mdevaev, 2020-05-07:
Oh, here's the thing :D I think I should add a description of this case to the documentation. I didn't think it would cause any problems. Thanks!

closed 2020-05-07 by stsirakidis

mdevaev, 2020-05-07:
By the way how about the quality slider problem? Is this gone too?

stsirakidis, 2020-05-07:
Regarding Kaspersky...the solution is to set the website of pikvm in network in the exclusion list (Protection -> Private browsing -> Categories and exclusions -> Exclusions). Afterwards uploading works like a charm!

With AV activated or not....I can downgrade the quality with the first attempt....upgrade the quality to a higher value may take 2 or 3 attempts...but it finally works.

mdevaev, 2020-05-07:
As for the quality - as I wrote, I should try it myself with your device. I need to look at requests, status, and so on. Otherwise, I can't help because the problem doesn't reproduce on any of my devices.

stsirakidis, 2020-05-07:
See please my PM to you.