pikvm/pikvm #18: Question: Can this be used as a serial port IPKVM rather and HDMI?

MrClayPole, 2020-05-10:


MrClayPole, 2020-05-10:
I'm looking for power management over the network via IPMI and via the web interface. Is this currently supported?

The question around the serial console rather than HDMI is all the hosts I'll be managing are linux without desktops so was just wondering if I could get away with the cheaper Pi ZeroW and serial console rather than Pi 4 & HDMI?

mdevaev, 2020-05-10:
Yes, you can control by IPMI. Power on, power off, and restart are supported. Ipmitool's requests will be translated into commands for the relay.

In fact, HDMI can be used even with Zero. But if you don't want to spend money on it, you can use the serial console without it. This will not be so convenient, because the project does not provide a special interface for the series console, except for the terminal emulator in the browser. But everything will work.

stsirakidis, 2020-05-10:
Btw, what is the "logic" with IPMI regarding power on or off...since the relay for the ATX power acts only on "bridging" the two PINS on the motherboard?

mdevaev, 2020-05-10:
Good question. To understand the current state, the server reads the power led. That is, you will need to connect it to the optocoupler (shown in the diagram). The off command will only be triggered when the led is on, and vice versa.

stsirakidis, 2020-05-10:
Cool!

Btw, pls edit your documentation regarding the passwords...the should not contain the # sign...the rest of the password after the "hash" sign will be implemented as comment.... :-)

mdevaev, 2020-05-10:
What file are you talking about?

stsirakidis, 2020-05-10:
````
# For SSH root user
ROOT_PASSWD = root
````

e.g. setting the password to "root#blabla". The way that the scripts are building the docker images implement the # sign as comment, when the scripts run and substitute the password_variable with the passwords set in config.mk. If these passwords contain the # sign, then the passwords are not the same as when the SD card is build. Sorry...I should have open a new issue/thread therefore...

mdevaev, 2020-05-10:
Don't worry. I will see how best to fix this or document this feature.

MrClayPole, 2020-05-10:
I want to managed a number of "odroid H2" systems using a bare metal deployment tool "MAAS". In order to turn systems on and off as required by MAAS, I need to use one of the supported methods. IPMI is a supported method. 

Thanks for your help with this I'll have a play with a Pi Zero/Pi 4 and Serial/HDMI and see which route I like the best.

closed 2020-05-10 by MrClayPole

mdevaev, 2020-05-10:
@stsirakidis try to use `ROOT_PASSWD = pass\#word`. It shoud works.