pikvm/pikvm #21: Add ability to mount writeable images

mallendeo, 2020-06-28:
It would be really nice if we could mount read/write images to use them as virtual hdds or usb drives, also set other parameters like GUID.

e.g: could be "writeable" or "read-only"

[image]

mdevaev, 2020-06-29:
To be honest this feature is already implemented but required some CLI actions to use. The reason is RO-filesystem. We have a separate partition on SD card to storing images. When user uploading the new image this partition comes to RW-mode, then switches back to RO. RO is necessary to pevent filesystem damage on power loss.

If we want to place the switch directly in the interface, the file system will have to be switched to write mode for a long time. Firstly, this will require a significant rework of the logic of work, and secondly, a frivolous user may just forget to switch everything back. So at the moment, I left only the possibility of using read-only images in the interface, because this is (among other things) the most frequent case, and images for writing can be connected via the console. In particular, there is a web terminal for this purpose. If you're interested in how this works, you can look at the `kvmd-otgmsd` command. I'll probably do what you ask in the future, but it's really a lot of work.

As for the interface, I really like the way it looks. I think I'll borrow an idea about the media type switch. The only thing is, I can't emulate HDD. This is limited by the current core capabilities. So for now, either CD-ROM or USB Flash.

mallendeo, 2020-06-29:
Thank you. I didn't know about the `kvmd-otgmsd` command. It seems to work, but it doesn't show up on the UI

`rw && kvmd-otgmsd --set-image /var/lib/kvmd/msd/images/unraid-usb-small.img --set-rw 1 --set-cdrom 0`
[image]

The `kvmd-otgmsd -m` (`kvmd-otgmsd -m | grep unraid`) command doesn't show the info of the current msd though. I think I can get around that for now.

mdevaev, 2020-06-30:
Please show the output of `journalctl -u kvmd` during this error.

mdevaev, 2020-06-30:
Okay, I think I reproduced the problem. "Offline" reports when you're disconnect drive. The log in that moment:
```
Jun 30 04:47:36 pikvm kvmd[328]: Traceback (most recent call last):
Jun 30 04:47:36 pikvm kvmd[328]:   File "/usr/lib/python3.8/site-packages/kvmd/plugins/msd/otg/__init__.py", line 437, in __reload_state
Jun 30 04:47:36 pikvm kvmd[328]:     raise MsdError("MSD has been switched to RW-mode manually")
Jun 30 04:47:36 pikvm kvmd[328]: kvmd.plugins.msd.MsdError: MSD has been switched to RW-mode manually
```
The reason is that the KVMD is quite paranoid in terms of tracking drive state. If you'he made some changes that do not cover its internal logic (for example, RW-mode), the system believes that it is better to stop working than to do something incomprehensible with possible disastrous consequences. Just switch the drive to RO mode and the interface will work again: `kvmd-otgmsd --set-rw=0`.

You should also know that the `rw`  ommand does not switch the partition to `/var/lib/kvmd`. It only applies to the basic functions of the operating system. For remounting use command `kvmd-helper-otgmsd-remount`. In addition, another useful command: `kvmd-helper-otgmsd-unlock`. It unlocks usb driver if the server's OS acquired it.

mallendeo, 2020-06-30:
Just checked and it is the same error I'm getting. I was running `mount -o remount,rw /dev/mmcblk0p3` at boot, so that might be the case. Nice to know about those commands, thanks.

connected201, 2021-05-20:
> 
> 
> Just checked and it is the same error I'm getting. I was running `mount -o remount,rw /dev/mmcblk0p3` at boot, so that might be the case. Nice to know about those commands, thanks.

Hi, you have solution about your use case?
best regards.

mdevaev, 2021-08-15:
Note just for me: https://www.spinics.net/lists/linux-usb/msg214103.html

connected201, 2021-08-17:
i tested and working very well, also i able to install linux on mounted .img file :)

mdevaev, 2022-07-23:
It's time.

mdevaev, 2022-07-24:
It's done and available since KVMD 3.126. Now you can upload the flash drive image to MSD, write some files to it or even install entire OS to the PiKVM MSD drive (I don't know why, but you can), and download the image back. Use the new feature wisely.

[image]

closed 2022-07-24 by mdevaev

connected201, 2022-07-25:
Great improvement.

sharicov, 2024-03-28:
Doesn't work for me, when connecting the image writable switches to Off automatically, only cli commands work:

kvmd-helper-otgmsd-remount rw
kvmd-otgmsd -i 0 --set-rw=1 --set-cdrom=0 --set-image=/var/lib/kvmd/msd/ventoy.img

mdevaev, 2024-03-28:
Switch type to flash firstly, then RW flag (in gui both).

sharicov, 2024-03-29:
That's exactly the order I'm doing it, rw switch to ro by itself when pressing connect.
BTW using the latest v2 version on zero 2 w

mdevaev, 2024-03-29:
Maybe something wrong with permissions. Try new image on another sd card.

sharicov, 2024-03-29:
Is there anything I can check in logs? I only installed fresh new system several days ago.

File permissions:
-rw-r--r-- 1 root root 19G Mar 29 04:58 ventoy.img

mdevaev, 2024-03-29:
:ok_hand:

sharicov, 2024-03-29:
BTW in the guides (https://docs.pikvm.org/msd/#multiple-drives and others) it says PiKVM should be restarted when switching mass storage type, or ro to rw, in order host PC to recognize the change, it's not a must, running USB Tree Viewer as admin and restarting the PiKVM device does the same on USB level.

mdevaev, 2024-03-29:
It depends on the host OS.

sharicov, 2024-03-29:
Sure, talking about Windows