pikvm/pikvm #22: Possibility to further use case LEDs and Buttons

fwillo, 2020-06-28:
Hi,

I found out about your project the last couple of days and I'm highly motivated to realize this for my own server at home!

In order to setup the whole as clean as possible I want to design a PCB, which will take one PCI slot (without connecting to PCI ofc) of the case and offers a generic connector for the ATX control and LEDs. However, the schematic breadboard design is not hinting any possibility of further using remaining buttons and LEDs of the case. 

Out of my (current) understanding, it should be done by simply wiring the buttons and LEDs to the respective parts  of the optocopplers. The resistor at Pin 1 of the LED optocopplers has to be adapted probably in order to permit enough current, I guess?

**My question (tl;dr):** Can you give some hints on how to further use remaining buttons/LEDs of the case?

Anyway thank you in advance for this neat project!

joBr99, 2020-06-28:
Hi fwillo,
i've not yet tested this yet, but for my understanding you only have to wire your case leds/button in parallel to the wires that you connect to your motherboard from the ATX control PCB.

mdevaev, 2020-06-29:
Hi. Usually a parallel connection is more than enough to use ATX. And yes, you're right, the resistor will give sufficient current. By the way, subscribe to the discord channel. Some guys are already doing something similar to what you have in mind. I'm sure you can do something cool together.

https://discord.gg/bpmXfz5

mdevaev, 2020-07-27:
I believe I have answered your question. If there are any problems - feel fry to open a new issue..

closed 2020-07-27 by mdevaev