pikvm/pikvm #26: Feautre request: FPGA capture device

axet, 2020-07-27:
Hello!

A little discovery shows it is possible to create FPGA based capture hardware device using simple fpga boards and VGA output.

If you google you will find a lot of FPGA VGA examples and ready verilog scripts to create simple output -> VGA, but simple rework of code can create perfect and super cheap FPGA capturing device using standard VGA output.

How do you like this idea?

mdevaev, 2020-07-27:
Hi! We will probably do something similar in one of our own boards. For now, I recommend to use a VGA-HDMI converter (the link is in the README). It's not the best solution, but it's better than nothing.

Right now we are making a v3 platform, but there is no room for vga on it. it will probably be either in the expander Board (which will turn v3 into a multiport version), or in a hypothetical v4.

I would appreciate it if you could provide links. I think we can use them.

axet, 2020-07-27:
Sure. I found nice video about fpga hardware and github repos with required information. Quite detailed. But I was lazy to order stuff from china, and did not progress further. Here is what I found:

Basic video about tinyfpga + vga, speed, CPU limitations (basic), basic VGA logic, and actual working device (vga starts from 9:20):

  * https://www.youtube.com/watch?v=ZNunxg7o8l0

Lets build VGA card (must watch, very detailed about vga):

  * https://www.youtube.com/watch?v=l7rce6IQDWs

Arduino VGA beginer:

  * https://www.youtube.com/watch?v=qJ68fRff5_k

Then, you gonna need VHDL vga libraries:

  * https://github.com/juanmard/screen-pong