pikvm/pikvm #27: orange pi build

centralhardware, 2020-07-28:
can i build pikvm for board orange pi pc2

mdevaev, 2020-07-28:
For Orange Pi will be required to modify software to support GPIO. It's not difficult and I can do it. The second difficulty is that you will need to use manual installation, because all the build scripts run on Raspberry. You will also most likely have to work with USB HDMI, since TC358743 doesn't seem to work with Orange.

PS: Если вы разбираетесь в оранжах и настроены серьезно, я готов помочь.

denisio, 2020-08-04:
I would like to port to Armbian Orange Pi Zero (OTG).
How to emulate keyboard and mouse with OTG?

Я хотел бы портировать на Orange Pi Zero, 
как у вас работает эмуляция клавиатуры и мыши (есть OTG) ? 
Через самописные модули ядра?

Cherkes001, 2020-08-04:
I join the question of porting to Orange PI, I would like to be able to run this useful thing on a Pi PC, it is cheaper than a Raspberry Pi and has an OTG connector on board like the same RPi 0/4, but since you already wrote about video capture, USB-HDMI is most likely the only way out. Not so long ago, interesting options appeared, for example - [Ali](https://aliexpress.ru/item/4000917130635.html) .
Our OPi community can be found in telegram - https://t.me/opi_sbc

Присоединяюсь к вопросу о портировании на Orange PI, я бы хотел бы иметь возможность запускать эту полезную вещь на Pi PC, он дешевле чем Raspberry Pi и имеет на борту OTG разъем как те же RPi 0/4, ну а так как вы уже написали  насчет захвата видео, USB-HDMI - вероятнее всего единственный выход. Не так давно появились интересные варианты, например - [Ali](https://aliexpress.ru/item/4000917130635.html) .
Наше комьюнити OPi можно найти в телеграм - https://t.me/opi_sbc

mdevaev, 2020-08-04:
OTG works via a regular libcomposite module.
For GPIO, I can make the necessary patch.
I think the only difficulty is in automating the installation. But it seems that Arch Linux ARM can be used on Orange. You will probably only need to build the packages (you can do it directly on Orange) and just install them.

centralhardware, 2020-08-10:
What is required now from those interested in this port?

Cherkes001, 2020-08-10:
> What is required now from those interested in this port?

Availability of usb-hdmi
Try to deploy the software and identify the resulting errors
Then we need to solve the work of the OTG so that we can abandon the arduino

mdevaev, 2020-09-15:
If someone is still interested in porting to Orange Pi, I am currently working on porting ATX code from RPi.GPIO to libgpiod. Now you don't have to write a separate driver for Orange Pi GPIO and other boards, since libgpiod works everywhere.

https://github.com/pikvm/pikvm/issues/65

arpbne, 2020-10-01:
[image]
Just to show approbation for idea, thats how orange pi zero looks with hdmi grabber attached.
Of course toshiba chip won't work (afaik csi != mipi csi), so we have to use hdmi grabbers.
Few months ago interesting device appeared, that kills other competitors in that price range.
It is sold for example as "rullz hdmi grabber" on aliexpress for about 10$.
It is not some hdmi->av->usb/uvc converter - its custom ASIC chip (ms2109) that does the HDMI-UVC conversion and it does it well (eg, resolution change is supported and overall experience is good - not for streaming games I guess, but for kvm-over-ip it will works well).

Raspberry pi 4 + TC chip works super good, but in most cases it is total overkill and uneccesary waste of money.
In that case, orange pi + ms2109 looks very promising.

Informatic, 2020-10-04:
@arpbne what a nice coincidence to see my own photo lurking around the issues. Makes me wonder how you actually managed to get that? :)

That's NanoPi Neo with pretty much fully working kvmd/ustreamer setup based on Yocto instead of Arch Linux with Macrosilicon MS2109 capture card wired directly in.

When I started that project a ~month ago I intended to rewrite gpio support to libgpiod, but after leaving it for a couple of weeks untouched I've been greatly surprised with @mdevaev work implementing that.

I'll release a yocto meta-layer with relevant packages / configuration soon.

mdevaev, 2020-10-04:
@arpbne Cool! But MS2109 has a number of problems. Sometimes this may spontaneously turn off or not work in the BIOS.

@Informatic do you want me to do the polling mode?

Informatic, 2020-10-04:
Interrupt-capable pins are present on all Allwinner H3 boards I've seen so far, so this isn't much of a blocker, if anyone is actually willing to use it on unsupported board. (and they'll most probably happen to find above note in case they stumble upon said error) I assume you might have some more important tasks to do anyway.

sisland, 2020-12-02:
Orange Pi owner here willing to help. I own Orange Pi Zero H3 (256mb) and Orange Pi Zero Plus 512mb (H5) [http://www.orangepi.org/OrangePiZeroPlus/](url)

Let me know if I can help testing something.

wnark, 2021-03-11:
> OTG works via a regular libcomposite module.
> For GPIO, I can make the necessary patch.
> I think the only difficulty is in automating the installation. But it seems that Arch Linux ARM can be used on [...]

Hello, I have a few questions.
1. Which version of the software package in https://pikvm.org/repos/ can I install directly on orangepi archlinux?
2. What files do I need to modify so that pikvm can use OTG usb and ms2109.
Because the performance of orange pi zero is higher than that of raspberry pi zero, it can also work normally on USB 2.0 through power consumption control. I have already bought one and want to see if it will be smoother than raspberry pi zero.
Thank you.

ketsa, 2022-02-28:
Lack of available raspi makes this a need.
Support more boards please.

xe5700, 2022-07-22:
Try my project install pikvm on non rpi device.
https://github.com/xe5700/kvmd-armbian