pikvm/pikvm #30: ComputeModule

KuDeSnik33ra, 2020-07-29:
RaspberryPi with hat is very big and difficult to cool, so, why don't you think about making board for RPi ComputeModule, or JetsonNano CM with some cooling solutions - it will be very handful. It could be in form of PCI card - physically only, not using PCI-connector - to get power inside pc and connect RJ-45 from back of case.

mdevaev, 2020-07-29:
Hi! The problem is that there is no Compute Module 4, and previous versions do not support the OTG required for Mass Storage emulation. Plus, it's a much larger circuit board in size and complexity. Using RPi4 is a compromise between simplicity and cheapness. This is our first product, but in the future we will try to do something more complex.

JsBergbau, 2020-07-30:
Are there cheap, lets say "IO boards" where you can plug the compute module in and get powersupply, LAN connection and so on?
The RPI is also often small enough to get it into PC case somewhere. If you have cooling problems, try this case https://www.berrybase.de/neu/armor-geh-228-use-f-252-r-raspberry-pi-4-schwarz its passiv but still really powerful.

KuDeSnik33ra, 2020-07-30:
I am speeking about future v3 hardware, using HAT board make such cooling cases not usable. Developing carrier board for CM is preffered, but, yes, CM4 not released yet, and there are no alternatives in capabilities and cost. 

Storing raspberry inside PC make problem to connect it - you will need non-disconnectable wires going inside case.

KuDeSnik33ra, 2020-07-30:
Isn't CM3+ capable to emulate massstorage OTG?

KuDeSnik33ra, 2020-07-30:
Maybe this? https://www.arrow.com/en/products/900-13448-0020-000/nvidia

KuDeSnik33ra, 2020-07-30:
You can use as a start this carrier https://github.com/antmicro/jetson-nano-baseboard

mdevaev, 2020-07-30:
We expect that many users already have RPi4. Your options are good, but they mean that we will have to throw away everything we have already done and give up RPi4 users. We are not a large company, so we can't afford it at this stage.

KuDeSnik33ra, 2020-07-30:
Why throw away anything? Do both. Start kickstarter company maybe, hire up new developers.

mdevaev, 2020-07-30:
Right now, we don't have the resources to do this. One person writes software, one person writes hardware. After we sell the kits on kickstarter, we will get the funds to hire people and do more interesting things as you suggest.

coreylista, 2020-08-01:
> RaspberryPi with hat is very big and difficult to cool, so, why don't you think about making board for RPi ComputeModule, or JetsonNano CM with some cooling solutions - it will be very handful. It [...]

Is something like this what you are asking about?

I am currently working on a PCI holder to 3D print for my RiPi.. 
[image] 
[image]

I have shared the files on the discord chat... I am still refining it personally. Others have used it to create their own version, you can find them in the discord chat.

mdevaev, 2020-09-05:
Okay, I guess there's nothing more to discuss at this stage. Feel free to reopen if needed.

closed 2020-09-05 by mdevaev

NiKiZe, 2021-10-04:
Here is another vote for CM4 board, I think it can reduce complexity compared to the current RPi4 HAT.
You can probably also get it to a smaller footprint. So in the _future_ I do look forward to a CM4 based version as well.

wnark, 2021-10-04:
> Here is another vote for CM4 board, I think it can reduce complexity compared to the current RPi4 HAT. You can probably also get it to a smaller footprint. So in the _future_ I do look forward to a CM4 based version as well.

I tried it, but the capture card used by CM4 cannot be driven normally under pikvm, mainly because the csi-4 driver requires a newer Raspberry Pi kernel, use
`dtoverlay=tc358743,4lane=1`

[image]

JsBergbau, 2021-10-04:
> RaspberryPi with hat is very big and difficult to cool

Since CM4 module uses the same CPU as Raspberry PI4, should there be any difference in cooling? It should produce the same heat and for the compute module to drive you need other hardware, which is much more complex and of course costs more. I like the idea of using the Raspberry PI4 because it is such a widely used hardware and easily available and developing a version for CM4 and RPi 4 brings additional costs which could be saved by focusing solely on RPI4 (and Zero W).

mdevaev, 2021-10-04:
We're working on it.

wnark, 2021-10-05:
> We're working on it.

> We're working on it.

According to the actual measurement, the micro USB can successfully simulate the device,
but the HDMI-CSI4 & HDCP of Dongmu Electronics cannot be successfully driven.
Even if they are all TC358743G, it may be because HDCP uses a special certification chip circuit, which may require a Blu-ray player to be recognized normally, but I don't have it. So, don’t test the HDMI-CSI & HDCP version for now
I will use Beili Electronics to give it a try next.
[image]
[image]
[image]

ishioni, 2021-11-01:
POE powered version a possibility as well?

JsBergbau, 2021-11-01:
There is now the Raspberry PI Zero 2W available which has a quadcore 1 GHz CPU with 64 Bit support (ARM Cortex A53) for 15$. This device could be exactly what we need. Little powerconsumption, but still very powerful. https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/