pikvm/pikvm #32: Documentation change request

mr-gosh, 2020-08-04:
decompression is done by etcher itself - ne need to decompress before flashing ;)

mdevaev, 2020-08-07:
I just tested this and can't recommend it for use.
* This works very slowly. Much slower than if you first uncompress and then flash the image. At least on my laptop. Perhaps this problem can occur in other people.
* While running in this mode, Balena Etcher consumes a huge amount of RAM (approximately 5 gigabytes in peak). This may cause the system to freeze.

mdevaev, 2020-08-07:
The new text:
> Decomress the image using your File Manager. If you don't have an archiver (on Windows for example) - the [7-Zip](https://www.7-zip.org) is a great choice. Please note: Balena Etcher have ability flash compressed images, but it is much slower and the process consumes a huge amount of RAM. We strongly recommend you to decompress the image first.

Thanks for the issue!

closed 2020-08-07 by mdevaev

mr-gosh, 2020-08-07:
yeah - you are right - its much faster with decompression before - perhaps you should remove my suggestion again - sorry.