pikvm/pikvm #33: How-To: Additional Info

Marthisdil, 2020-08-12:
For those random folks who come across this from Reddit, Google, etc, you should have a section telling folks on how to connect things up to the computer/server they are looking to remotely manage.  Would be helpful.

Also a section with commentary about options for older systems that have, say, VGA or DVI output on the motherboard, but no HDMI, etc.

Little touches to the documentation :)

Nice work!

mdevaev, 2020-08-12:
Thanks, good point. I will.

tryallthethings, 2020-08-14:
On the topic of documentation: Any reason why you're not using the GitHub wiki for this?

mdevaev, 2020-08-14:
Historical legacy :)

tryallthethings, 2020-08-14:
Maybe it's a good time to switch to the wiki with this new project. Extensive documentation is easier to navigate / organize within a wiki. I would like to contribute some docs as I'm diving into pikvm.

mdevaev, 2020-08-15:
I haven't edited a wiki on github in a very long time. Do the user need a pull request for changes now?

tryallthethings, 2020-08-15:
Yes, kind of. Although GitHub by default only has two permission settings for wikis (public edit access or project collaborators) there are ways around this to enable pull requests for wiki edits: For example [this](https://www.growingwiththeweb.com/2016/07/enabling-pull-requests-on-github-wikis.html)

mdevaev, 2020-08-17:
I suggest that we continue in the main repository for the time being. At some point, I will gather my strength and transfer everything to the wiki.

tryallthethings, 2020-08-17:
Sure. I already submitted a pull request to the documentation yesterday. I would recommend moving a the documentation / FAQ from the landing page to a separate page though. It grew quite a lot and is getting hard to navigate / find information.

mdevaev, 2020-08-17:
I will move some of the documentation to other pages, but I want to leave the main instructions here, because this way it is perfectly indexed by keywords in google. At least at this stage of the project.

Kaydron1000, 2021-04-09:
I was digging into the docs here and found out a lot of hidden items in the pages folder. I think it would be a lot easier for people to navigate on a wiki or a github project page.

I first queried in Discord and there seemed to be interest in some updated documentation. I threw together a quick example of documentation a little more user friendly here: https://kaydron1000.github.io/SampleDocs

The theme is decent enough to start with. Newer themes or customization can be done easy enough in the future after the pages are made.

Let me know I would be happy to pull and update the docs to something like what I linked.

mdevaev, 2021-04-09:
This is nice, but I would prefer to keep a step-by-step, like, first the project description, then the build instructions, and then the general text about what to do next.

Kaydron1000, 2021-04-10:
I am reading through all these docs and there is A LOT of options to change (uncommenting lines, commenting lines, adding lines, or deleting  lines) around depending on hardware or use case. I'm assuming there is the blue screen shell menu available in arch as there is in debian? this probably should be a new issues/request feature if similar "easy" to use menu scripts should be created for users to change config. The menu scripts would eliminate the "rw" and "ro" commands that i'm sure many people initially get confused by.

mdevaev, 2021-04-11:
That would be great, but I have absolutely no time to do it, as there are more pressing problems such as H264 that no one but me will solve.

Kaydron1000, 2021-04-11:
Np I was going to tackle a first run of updating the documentation then possibly poke around in a shell script menu. I made a simple one before, but I see it can be very useful here.

mdevaev, 2021-04-13:
The documentation has been updated. Unrelated things should be in a separate issue

closed 2021-04-13 by mdevaev