pikvm/pikvm #34: Access pikvm over the internet

tryallthethings, 2020-08-12:
First off: Awesome project. From what I've read this is exactly what I was looking for.

Do you have any plans on implementing access to pikvm over the internet? I was thinking of something like an automatic VPN connection once a network with internet access is detected or a server that pikvm is connecting to that provided the access (similar to what TeamViewer does).
My idea is to ship this to friends / family, let them hook it up to the PC / laptop and get access to their machine without having the trouble of any port forwarding.

mdevaev, 2020-08-12:
Hi and thanks)

Right now we're providing the [tailscale](https://tailscale.com) binary for this. After installation the OS you can install tailscale from our repo (`rw; pacman -S tailscale` and follow the instructions) and you will give access to pikvm over internet without external IP address.

mdevaev, 2020-08-12:
Here the manual:
1) `rw; pacman -S tailscale`
2) `systemctl enable --now tailscaled`
3) Follow instructions https://tailscale.com/kb/1036/install-arch from the step 3.
4) Run `ro` command and `reboot` then.

tryallthethings, 2020-08-12:
Never heard of tailscale so far. But from a first look it seems to be exactly what I'm looking for. Will give it a shot once my RPi 4 arrives. 

Thank you!

mdevaev, 2020-08-13:
Added some documentation about it: https://github.com/pikvm/pikvm#access-to-pi-kvm-from-the-internet

closed 2020-08-13 by mdevaev

tryallthethings, 2020-08-15:
Just a quick follow-up: Tailscale works absolutely amazing. Starts on device boot-up and if there's any network connection (regardless of wired or wireless) it will connect. Also works just fine with WLAN roaming and switching between wired and wireless connection.