pikvm/pikvm #36: Can't build for RPI2

ech1965, 2020-08-14:
Dear,

I'm trying to build an image for my RPI2 + HDMI CSI Hardware.
I followed instructions in building_os.md but build fails:

`Step 21/77 : RUN gcc -fPIC -shared -o /usr/local/lib/libpreload-semop.so /tmp/semop-wrapper.c
 ---> Running in 5cfe33b6374d
Allocating guest commpage: Operation not permitted
The command '/bin/sh -c gcc -fPIC -shared -o /usr/local/lib/libpreload-semop.so /tmp/semop-wrapper.c' returned a non-zero code: 1
make[1]: *** [Makefile:209 : os] Erreur 1
make[1] : on quitte le répertoire « /home/tsos/os/.pi-builder »
make: *** [Makefile:53 : os] Erreur 2
`

Freshly installed Ubuntu 20.04 server in a Virtualbox VM ( macos host)

Please help !
Etienne

mdevaev, 2020-08-15:
For the protocol: the QEMU version is fixed for 5.0.0. Let's continue discussing the other issues in discord.

ech1965, 2020-08-15:
Bug reported upstream: https://bugs.launchpad.net/qemu/+bug/1891748

@mdevaev : Thanks a lot for your help preparing the reproduction steps.

mdevaev, 2021-07-30:
Related: https://bugs.launchpad.net/qemu/+bug/1880225

Still not fixed but at least we have workaround.

mdevaev, 2021-07-30:
https://patchwork.ozlabs.org/project/qemu-devel/patch/1273756645-11149-1-git-send-email-Martin.Jansa@gmail.com/

mdevaev, 2021-08-28:
TODO: report here: https://gitlab.com/qemu-project/qemu/-/issues

mdevaev, 2022-01-20:
Fixed in dev branch, will be available in QEMU 7.0

closed 2022-03-21 by mdevaev