pikvm/pikvm #39: Feature request: Keep mass storage dropdown open

tryallthethings, 2020-08-15:
While uploading a new image the mass storage dropdown menu disappears after selecting the .iso file from the computer with a double click. It would be great if either the upload would start immediately after selecting the image or if the dropdown would stay open to click "Start". It does stay open if the file is selected in the explorer window and confirmed with "Open".
Same goes for "Disconnect drive". Users might want to swap to a different image but the dropdown menu disappears as soon as you click on "Disconnect drive" and you'll have to reopen it again.

mdevaev, 2020-08-15:
> It does stay open if the file is selected in the explorer window and confirmed with "Open".

The menu shows what you have selected and shows how much space the image takes up. It is assumed that you should check whether you made a mistake with the choice and only then click upload. It seems to me that in combination with PR with button renaming, automatic loading no longer makes sense.

> Users might want to swap to a different image

Good point.

tryallthethings, 2020-08-16:
Sorry, I probably explained it poorly. 

Case 1: Selecting image via clicking "Open" - works great as the mass storage dropdown stays open.
[image]

Case 2: Selecting the image with mouse double-click - not as great as the mass storage dropdown is closed.
[image]

mdevaev, 2020-08-16:
From what I've found out, this is a Windows file dialog bug that produces an additional mouse event. This is reproduced in Chrome and Firefox on Windows, but not on other systems. Proof: https://stackoverflow.com/questions/25893328/avoid-mouseup-event-when-selecting-a-file-from-dialog-when-double-clicking

In other words, the window generates a mouse event and then the web interface interprets it as a click outside the menu. If you move the file dialog so that the mouse click is above the menu, it will not close.

Given that the window and event management code in the interface is extremely complex, I would prefer not to work around this bug. This may break some other behavior, since all solutions to the problem involve inserting additional conditions into the event handling code (which is already very complex).

tryallthethings, 2020-08-16:
Fair enough. It's a minor inconvenience.
Maybe you move the mass storage configuration to a floating window like the stream or keyboard at some point which would work around that bug.

mdevaev, 2020-08-16:
Perhaps in the future. But not in the current interface concept.

mdevaev, 2020-08-16:
PS: I did the menu is not closed after the connection and disconnection of the drive.

closed 2020-08-16 by mdevaev

tryallthethings, 2020-08-16:
Thanks! Did you invent time travel? :smile:

mdevaev, 2020-08-16:
Ah, if only.