pikvm/pikvm #40: Feature request: Web interface translation

tryallthethings, 2020-08-15:
Translating the web interface would make it more accessible for a broader audience. 

From what I've understood you're using [pug formerly Jade](https://pugjs.org/) to generate the web interface. Javascript isn't exactly my forte but I found some information:
- https://github.com/pugjs/pug/issues/2922
- https://i18next.github.io/i18next/node/pages/doc_express.html

mdevaev, 2020-08-15:
It doesn't work. Pug is used for local generation of html files, the entire web interface is completely built on static html. I don't use node or any JS frameworks. That's why, in general, the interface is so fast :) But with localization of course because of this problem. Given that the entire IT users speaks english (even my russian users speaks english), this task will be done last by me, or with the help of those who need it most.

tryallthethings, 2020-08-15:
> That's why, in general, the interface is so fast
Yes and I love it. No waiting for nothing :smile: 
> Given that the entire IT users speaks english
True. I don't need it in my native tongue but I'm sure there will be requests sooner or later. Especially now that the project is gaining traction. At least we started the discussion and someone familiar with pug chimes in and works on this.

tryallthethings, 2020-09-18:
I understood that the web pages are static. So we'd either have to generate multiple versions of each page for each translated language (which is probably a pain in the ass to maintain) or do a client side translation with JavaScript.
Would [this - requires jQuery](https://github.com/wikimedia/jquery.i18n) or [this - same, but without jQuery dependency](https://github.com/wikimedia/banana-i18n) be feasible?
Seems rather easy to implement and maintain as each language can have their own translation file and be updated individually from the main release.

mdevaev, 2020-09-18:
I would prefer to do without any frameworks. At least on the client side. I did not suffer and wrote a light interface to then drag in a thick bloatware :) It seems to me that the problem can be solved in some other way. For example, if all texts are assigned by JavaScript when calling a separate function (as is customary in Qt with tr/translateUi).

I see several stages of solving this problem:
1. All text from HTML should be moved in JS.
2. The content of text elements must change dynamically [like this](https://wiki.qt.io/How_to_create_a_multi_language_application).

tryallthethings, 2020-09-18:
Can't say if the banana framework is thick or not. But I think the same functionality the framework offers can be achieved with our own implementation, as it just does a search and replace of text. Either by using English as a base language and translating text strings or using placeholders which get replaced by the text of the selected language. 
1. Depends on the implementation. It could either stay where it is with the proposal above and be replaced with placeholder-tags.
2. If we're doing client-side translation we could either reload the page or just the translation function to update the text.

wnark, 2021-03-08:
> It doesn't work. Pug is used for local generation of html files, the entire web interface is completely built on static html. I don't use node or any JS frameworks. That's why, in general, the [...]

hello， I am going to translate the webpage manually
The html file I found is this `https://github.com/pikvm/kvmd/blob/eb379296f2b3344bccc28ba4bac95554779ad554/web/kvm/index.html`

`\v2-hdmi-zerow\1\usr\share\kvmd\web\kvm`

Can I use the translated files when compiling?
I don’t know much about the compilation process, and I’m not sure where to modify the GitHub library after my fork.

thanks

wnark, 2021-03-22:
Hello, I encountered a problem when translating a webpage. If there is the following prompt, does it mean that the hardware (keyboard, mouse or capture card) driver is abnormal?
Or does it mean that the hardware of the Raspberry Pi is damaged?
```
                      <td><sup style="line-height:1">This is not a drill! A red icon indicates a current issue,<br>
                          a yellow one that was observed since the device booted up</sup></td>
```

wnark, 2021-03-22:
I am going to upload it to GitHub after the manual translation is completed, so that others can overwrite the files in the TF card (because I am not sure about the compilation process)

mdevaev, 2021-03-22:
This is information about a power shortage or overheating. The iron is not damaged yet, it may be soon.