pikvm/pikvm #41: Feature request: Capture and / or decode PC speaker signal

tryallthethings, 2020-08-15:
For a permanent installation the available options are already great (Power switch, Reset, HDD Led, Power LED) but is IMHO missing one crucial part. The internal PC speaker output for BIOS POST diagnostics. It would be a great addition to the already available information and control options in pikvm.
I'm seeing two options here:
1. Capture the audio and redirect it to the web interface
2. Decode the signal and display the beeps as some kind of morse code history in the web interface

I did some research and the PC speaker (piezo) seems to be controlled via voltage. (Wikipedia)[https://en.wikipedia.org/wiki/PC_speaker#Pulse-width_modulation]
In theory it should be possible to detect those voltages and the length. I've ordered a MCP3008-I/P and give it a shot once it arrives.

tryallthethings, 2020-08-18:
Did some thinking / research on this. Aren't you already doing something similar for the HDD / Power LEDs? Wouldn't adding another channel for the speaker be an option?

mdevaev, 2020-08-18:
In principle, I can, but what's the point? If you need signals, it is highly likely that your motherboard does not load and KVM will not fix this problem, since it is hardware-based. In other cases, the speaker is unnecessary.

Plus, now the status of LEDs is monitored not by interrupts, but simply by checking every 100 milliseconds. And it would also be great for signals to make a history to check them.

I consider this feature a minor one because I don't see much point in it. However, I will not refuse it.

tryallthethings, 2020-08-18:
There's more to the beep codes than just beeps == error and not booting. They indicate a successful POST on startup / reboot, notify about CPU overheating issues, if no keyboard is detected, CPU or system cooler not running properly or failed and many other things.
Some of those issues can be "fixed" via KVM (shutting down system on overheating / cooler failure, restarting pikvm to fix keyboard issue, etc.) 
For the other issues it could be great to know what the beep code is before going / driving to the pikvm controlled machine so you can already bring spare parts.

The signal history for LEDs / speaker would certainly be a great addition to diagnose issues.

maxpain, 2023-01-11:
Do you guys have an idea how to capture POST codes?

mdevaev, 2023-01-11:
It is impossible without aditional hardware.

maxpain, 2023-01-11:
According [to this article](https://en.wikipedia.org/wiki/POST_card) we can use USB:
> Another option are USB POST cards such as [AMI Debug Rx](https://ami.com/en/tech-blog/from-the-ami-archives-amidebug-rx-usb-debug-tool/), which use the [debug port common on USB 2.0 EHCI controllers](https://www.coreboot.org/EHCI_Debug_Port).[[1]](https://en.wikipedia.org/wiki/POST_card#cite_note-aptio-1)

maxpain, 2023-01-11:
[image]

rafimaryudwika, 2024-05-31:
According to this [repo](https://github.com/MrGreensWorkshop/RasPiPicoSDK-PicoBiosPostCodeReader?tab=readme-ov-file), it's mostly possible to implement POST code sniffing in PiKVM with proper LPC connector from PC (usually used for TPM) and GPIO from Raspberry Pi while it's quite risky.