pikvm/pikvm #43: WIFI_ESSID fails if the ESSID has spaces

bbeaudoin, 2020-08-18:
My wireless network ESSID is "GET OFF MY LAN". When configuring Arch Linux wireless using wifi-menu in rw mode, it creates an interface that netctl can enable; when escaped by `systemd-escape` the output, and device-specific unit file, appears as

~~~
wlan0\x2dGET\x20OFF\x20MY\x20LAN
~~~

During the build process ("Building OS"), `make os` calls `docker build` with the `WIFI_ESSID='GET \` resulting in

~~~
/bin/sh: 12: Syntax error: Unterminated quoted string
~~~

Looks like `BUILD_OPTS` of `~/os/.pi-builder/Makefile` (line 35) may be splitting based on the IFS, ignoring the quotes around the variable name. Unfortunately, and admittedly, it's been a while since I've coded a Makefile, for now I've created an alternate ESSID without special characters or spaces and was able to complete the build successfully.

nohajc, 2020-08-26:
@bbeaudoin I had a similar problem (wifi password containing spaces) and I've solved it by shell escaping (`\ `) in the Makefile.

mdevaev, 2021-01-16:
Fixed here: https://github.com/pikvm/os/pull/10

closed 2021-01-16 by mdevaev