pikvm/pikvm #44: API documentation?

codinghipster, 2020-08-19:
I'm looking for documentation, or even just the source code, for the API for pikvm. I looked through this repo, the website, and subreddit and I couldn't find anything.

If documentation doesn't exist I'm absolutely willing to help with that too..

tryallthethings, 2020-08-19:
Documentation doesn't exist (yet) as far as I know but you can find the source code here: https://github.com/pikvm/kvmd

mdevaev, 2020-08-20:
I would be very grateful if you could help with the description. You can find all the APIs here: https://github.com/pikvm/kvmd/tree/master/kvmd/apps/kvmd/api

mdevaev, 2020-09-04:
@codinghipster if you have any questions, I am ready to help you

codinghipster, 2020-09-04:
Thanks, I'm going to look into swagger or some other form of self-documenting library. I've never set that up with aiohttp so it should be interesting.

mdevaev, 2020-09-17:
Just started this: https://github.com/pikvm/pikvm/blob/master/pages/api.md