pikvm/pikvm #46: Alternative for 390 Ohm resistors.

deepwather, 2020-08-20:
Hello mdevaev,
I would like to build the v2 version of your project for my server. But I can't get 390 Ohm resistors in Switzerland.

I have already ordered: 10 Ohm, 100 Ohm, 220 Ohm, 330 Ohm, and 470 Ohm.

Can I use one or more of these instead of the 390 Ohm?

Friendly greetings
Michael Reber

mdevaev, 2020-08-20:
Hi. You can use 470 or 330.

deepwather, 2020-08-20:
Thanks for the quick response, so here's what I need:

For ATX control:
4x MOSFET relays OMRON G3VM-61A1.
4x 330 Ohm resistors.
2x 4.7k ohm resistors.

What are the advantages or disadvantages if I use the bigger resistor? or are there none?

Thanks

mdevaev, 2020-08-20:
The difference can only be noticeable in certain specific conditions. I think everything should work fine.

mdevaev, 2020-08-20:
Feel free to reopen if needed.

closed 2020-08-20 by mdevaev