pikvm/pikvm #47: Quality of Life improvements to update / restart pikvm

tryallthethings, 2020-08-20:
nice to have; low priority

- System and / or pikvm component update through web interface
- pikvm restart (system and / or daemon) via web interface
- update alias (.bashrc) or bash script similar to /usr/local/bin/rw to install the latest updates via pacman

mdevaev, 2020-08-24:
> pikvm restart (system and / or daemon) via web interface

Why is this necessary?

tryallthethings, 2020-08-24:
For example if an update requires a restart or in case of any issues.

n1nj4888, 2020-09-11:
I have a hardware HDMI kvm switch which, when the switch port is changed to another host, pikvm does not show any video (No Signal) - In this case, if I then reboot the pikvm then shows the video fine... So at least in this scenario, having the option to reboot pikvm itself via the web gui would  be better rather than then having to log in to SSH to reboot every time I switch hosts. 

Obviously though, having a fix that correctly updated displayed the new host video on host switch would be preferred👍

mdevaev, 2020-09-16:
@n1nj4888 You don't have to use a reboot to do this. Try simply reloading the HDMI EDID data: `/usr/bin/v4l2-ctl --device=/dev/kvmd-video --set-edid=file=/etc/kvmd/tc358743-edid.hex --fix-edid-checksums`

ndastur, 2020-11-07:
Is this what "Reset Stream" does? If not could it be added?

mdevaev, 2022-07-26:
Okay, what are we have today:
* Updating the OS is a complex operation that must be performed consciously, so it will not be done with one button as long as we use generic Linux OS and not something embedded like OpenWRT, or do not have a simple option to roll back all changes. For the same reason, I don't like the idea of command shortcuts: after the update, you need to do a reboot or an explicit ro.
* You can use [cmd gpio](https://docs.pikvm.org/gpio/#cmd) to pin command buttons like `reboot` and `poweroff`. Don't use it for update. So this is implemented already I think.

closed 2022-07-26 by mdevaev