pikvm/pikvm #50: Capture and record video

tuaris, 2020-08-21:
This is a feature request to be able to capture/record the video stream to a file on a SMB share.   This might already be possible using VNC and VideoLAN, but it would be nice to have a "record" button in the UI with a dialog choosing where  to save the video to a file server using SMB)

mdevaev, 2020-09-03:
I just want you to know that I have not forgotten about this request, I will deal with it as soon as possible, it is quite difficult to implement right now.

mdevaev, 2021-01-11:
Now you can record H264 videos. However, you need to use a web terminal to do this. Forwarding this function to the web interface requires a lot of effort and is a low-priority task. Now it is more important to replace MJPEG with H264 to reduce traffic. In the future, the problem with the interface may be solved.

https://github.com/pikvm/pikvm/blob/master/pages/video.md

closed 2021-02-09 by mdevaev