pikvm/pikvm #53: Can't build OS, wrong inside docker

arpbne, 2020-08-25:
Hi, I have problems running `make os`.
```
$ make os
[ ... ] 
Step 14/77 : RUN rm -f /etc/localtime 	&& ln -s "/usr/share/zoneinfo/$TIMEZONE" /etc/localtime
 ---> Using cache
 ---> 64e3d0ac22f9
Step 15/77 : RUN echo "HOOKS=(base udev block filesystems)" >> /etc/mkinitcpio.conf
 ---> Using cache
 ---> a9f9c08e6fdd
Step 16/77 : RUN echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist 	&& pacman-key --init 	&& pacman-key --populate archlinuxarm
 ---> Running in f2eafdce47d2
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
gpg: no ultimately trusted keys found
gpg: starting migration from earlier GnuPG versions
gpg: porting secret keys from '/etc/pacman.d/gnupg/secring.gpg' to gpg-agent
gpg: migration succeeded
gpg: Generating pacman keyring master key...
gpg: agent_genkey failed: End of file
gpg: key generation failed: End of file
gpg: Done
==> Updating trust database...
gpg: no need for a trustdb check
==> Appending keys from archlinuxarm.gpg...
gpg: key B55C5315DCD9EE1A was created 16088 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 16048 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 16048 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 16048 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A: no valid user IDs
gpg: key BC704E86B823CD25 was created 16088 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 16048 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 16048 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 16048 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25: no valid user IDs
gpg: key BF7EEF7A9C6B5765 was created 16088 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 16048 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 16048 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 16048 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765: no valid user IDs
gpg: key 77193F152BDBE6A6 was created 16088 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 16048 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 16048 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 16048 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6: no valid user IDs
==> Locally signing trusted keys in keyring...
  -> Locally signing key 69DD6C8FD314223E14362848BF7EEF7A9C6B5765...
==> ERROR: 69DD6C8FD314223E14362848BF7EEF7A9C6B5765 could not be locally signed.
  -> Locally signing key 02922214DE8981D14DC2ACABBC704E86B823CD25...
==> ERROR: 02922214DE8981D14DC2ACABBC704E86B823CD25 could not be locally signed.
  -> Locally signing key 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A...
==> ERROR: 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A could not be locally signed.
The command '/bin/sh -c echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist 	&& pacman-key --init 	&& pacman-key --populate archlinuxarm' returned a non-zero code: 1
Makefile:207: [makefile error ...]
```
When I add `RUN date` in the `.pi-builder/stages/os/Dockerfile.part` and run it again, it shows that date is set to 1970.
It worked once in my previous environment, now I'm running it it in qemu vm with debian buster and can't figure out whats wrong.
Why docker have such problems with setting date?

mdevaev, 2020-08-25:
This is because the container runs qemu for the arm build. One of the previous versions of glibc (your current perhaps) contains an error that caused qemu to work incorrectly with time functions. Use a VM with ubuntu 20.04 and everything will be fine.

arpbne, 2020-08-25:
Damn, that was quick 😁
I tried dozens of dirty hacks to overcome this, but no one of them worked. 
Gonna try that in a moment, thank you.

arpbne, 2020-08-25:
Yep, that was the case. Thanks Maxim )))

closed 2020-08-25 by arpbne

mdevaev, 2020-08-25:
Welcome