pikvm/pikvm #55: `ttyd login` doesn't work due to an Arch Linux ARM bug

bbeaudoin, 2020-08-29:
[https://archlinuxarm.org/forum/viewtopic.php?f=15&t=14716&p=64407#p64407](url)

At the moment, the "Terminal" functionality is a simple non-privileged shell. While this cannot be accessed without first authenticating to PiKVM and requires `su -` and a password to access full terminal functionality, there is a bug with the `login` program that ships with Arch Linux ARM in util-linux-2.36-2 preventing `ttyd` from being used with `login` directly.

I've documented two workarounds in the Arch Linux ARM bug reporting forum:

1. Use `strace ttyd login` instead of `ttyd login` in the systemd unit file or
2. Copy the `login` command from Raspbian for armv7l (32-bit) to `/usr/local/bin/` and use that version with kvmd

I'm testing the workaround without copying the foreign `login` binary from Raspbian using the first option. It's not perfect, here's what I'm presently seeing with the workaround in-place:

1. The terminal may fail to respond on initial boot
2. After `systemctl restart kvmd-webterm`, the `login: ` prompt seems to be consumed on initial connection and only the `password: ` prompt appears.

At present I don't recommend using this workaround if you need the terminal functionality and the current authentication options are sufficient. Here is my copy of `/etc/systemd/system/multi-user.target.wants/kvmd-webterm.service` I'm currently using, note that `login` requires root privilege to manage the session properly:

```
[Unit]
Description=Pi-KVM - Web terminal (ttyd)
After=network.target

[Service]
ExecStart=/usr/bin/strace /usr/bin/ttyd --interface=/run/kvmd/ttyd.sock --port=0 /usr/bin/login 2>/dev/null
Type=simple
Restart=always
User=root
Group=root

ExecStartPost=/bin/bash -c 'sleep 3; chmod 660 /run/kvmd/ttyd.sock; chown kvmd-webterm:kvmd-webterm /run/kvmd/ttyd.sock'

[Install]
WantedBy=multi-user.target
```

mdevaev, 2022-06-12:
Here some details: https://github.com/tsl0922/ttyd/issues/768

mdevaev, 2022-07-31:
Okay, I'm closing it because this is clearly a problem in ttyd and we will not use the login configuration in the OS image, even if it is fixed.

closed 2022-07-31 by mdevaev