pikvm/pikvm #56: new hdmi-csi board version

mpictor, 2020-09-06:
When I looked at the hdmi to csi adapter on alibaba, I noticed they suggested a new version https://www.aliexpress.com/item/4000152180240.html which has two sizes of ffc connector. If/when you know how it works, you might want to add info comparing them. In the photos of the new version, it appears that the smaller connector is rotated 180 degrees from its silkscreen and the solder pads which should hold it down, which is concerning...

You made it onto hackaday and from there into my news feed. That's some crazy soldering you've managed!

mdevaev, 2020-09-06:
This is the same device as described in the documentation. It just has a different form factor, the same chip TC358743 inside, and it works exactly the same. You can buy it and everything will work.

PS: Thanks :)

mdevaev, 2020-09-11:
I believe I answered your question :) Feel free to reopen and ask something if you need.

closed 2020-09-11 by mdevaev