pikvm/pikvm #57: Mouse Jiggler

maxexcloo, 2020-09-09:
Not sure if this is in scope but it would be awesome to have a mouse jiggle (or similar feature) to prevent a client computer from going to sleep/locking.

maxexcloo, 2020-09-09:
Potentially the macro feature could be used for this also?

mdevaev, 2020-09-09:
Hi! You're not the first user who asks to make this feature, and I will implement it. Now this is due to some technical difficulties due to the fact that pikvm uses an absolute mouse mode, not a relative one.

tryallthethings, 2020-09-10:
> Potentially the macro feature could be used for this also?
You could already use the macro function to achieve a similar functionality (although a bit hacky) if you need this functionality right now.
This script will press left CTRL every 15 minutes [script.txt](https://github.com/pikvm/pikvm/files/5203839/script.txt) for an hour. In theory, you could copy the block and repeat it to keep the machine awake for longer periods. And depending on your case you might want to use a different key such as the Num lock or an arrow key.

mdevaev, 2020-09-11:
In the next version, you will be able to loop macros: #63.

mdevaev, 2020-11-13:
Good news. With the advent of [relative mouse mode](https://www.reddit.com/r/pikvm/comments/js8wij/kvmd_24_the_relative_mouse_bluetooth_hid/), this is now possible.

tylersprice84, 2020-11-14:
This is good news! Thank you for your hard work.

mdevaev, 2021-02-24:
From #205 
> This might take some trial-and-error in terms of the correct process, but a way to instruct pikvm to keep a connected device awake, by either pressing a key or moving the mouse every X minutes. Obviously, the best solution would be one that doesn't interfere with actual use (you wouldn't want a keypress while in the middle of typing something, for example). Perhaps there is a way to detect when a connected session is idle or disconnected and then work to keep the connected device alive? The use case here is for when we're setting up new equipment that by default may go into sleep mode before we've had a chance to change those settings.

c0bra99, 2022-11-09:
> While we're waiting for native functionality to land in PiKVM, [pikvm-mouse-wiggle](https://github.com/jtyers/pikvm-mouse-wiggle) will do this for you. Disclaimer: I'm the maintainer.

@jtyers, Is this repository private? The link goes to a 404

jtyers, 2022-11-09:
Whoopsie, thanks @c0bra99, fixed

adamoutler, 2023-05-15:
I'm pretty sure this is able to be closed.  It would not be proper to add a Mouse Jiggler as an "official" feature and there are several available unofficial solutions.  

For example, this can be installed via WebTerm and requires nothing to be installed. 
[image]

The solution presented here provides an enable/disable button in the GPIO menu.  The solution uses absolute mouse movement once a minute and randomly selects a cursor location in the top left of your screen. The solution requires no externals and uses strictly bash-only built-in features (no external binaries). See this discord post for instructions on how to install.  https://discord.com/channels/580094191938437144/896216359963332628/1032353915561316352

JustinGuese, 2023-07-07:
why is it not proper? seems like a pretty normal functionality?

adamoutler, 2023-07-07:
> why is it not proper? seems like a pretty normal functionality?

Improper Mouse Jiggler Usage

Model: Default (GPT-3.5)

User
Can you reword this for me? 

> why is it not proper? seems like a pretty normal functionality?

Well, let's break it down.  Mouse jiggler is covered in most basic security classes. Mouse Jiggler functionality keeps the system from going to sleep every 5-15 minutes so that the user doesn't need to enter a password at a regular interval.  This is useful when performing work so that you don't need the user's  password or attempting to gain access to a system without permission so that you can maintain access.

 Mouse Jiggler subverts built-in operating system protection.  If you're a system administrator, the system already has a way to increase the timeout.  so this bypasses a system functionality and means this feature is a hacking tool.

It isn't appropriate to build hacking tools into a remote server/computer/laptop administration tool as presented out-of-box. However it is already within the capabilities of the PiKVM and therefore there is nothing stopping you from adding a Jiggler start/stop button to the GPIO menu.

ChatGPT
Why is it considered improper? It appears to be a fairly standard feature, doesn't it?

Let's analyze the situation. Mouse jiggler functionality is commonly taught in basic security classes. It prevents a system from entering sleep mode every 5-15 minutes, eliminating the need for the user to enter their password repeatedly. This is convenient for tasks that require continuous activity or when trying to maintain system access without authorization.

However, mouse jiggler undermines the built-in security measures of an operating system. If you are a system administrator, there are already ways to adjust the timeout settings without resorting to this method. Therefore, incorporating this feature into a remote server/computer/laptop administration tool, as it comes by default, is not appropriate since it can be seen as a hacking tool.

Nevertheless, the PiKVM already has the capability to incorporate a Jiggler start/stop button into the GPIO menu, so there is nothing preventing you from adding this functionality if desired.

adamoutler, 2023-07-07:
> why is it not proper? seems like a pretty normal functionality?

Let's analyze the situation. Mouse jiggler functionality is commonly taught in basic security classes. It prevents a system from entering sleep mode every 5-15 minutes, eliminating the need for the user to enter their password repeatedly. This is convenient for tasks that require continuous activity or when trying to maintain system access without authorization.

However, mouse jiggler undermines the built-in security measures of an operating system. If you are a system administrator, there are already ways to adjust the timeout settings without resorting to this method. Therefore, incorporating this feature into a remote server/computer/laptop administration tool, as it comes by default, is not appropriate since it can be seen as a hacking tool.

Nevertheless, the PiKVM already has the capability to incorporate a Jiggler start/stop button into the GPIO menu, so there is nothing preventing you from adding this functionality if desired.

JustinGuese, 2023-07-14:
Yes I can confirm that just recording a macro & looping it achieves exactly that. Would be nice to just have it in one click though, but still okay as is

mdevaev, 2023-10-23:
It's done in the last release. It performs short mouse movements every minute until you use PiKVM. Works with both mouse modes: absolute and relative. In addition, being enabled, it does not interfere with normal work with PiKVM: it will only interfere when you are not interacting with a virtual keyboard or mouse. The time of inactivity is counted from your last actions.

[image]

closed 2023-10-23 by mdevaev