pikvm/pikvm #58: Blurry Video

maxexcloo, 2020-09-09:
I'm using a Raspberry Pi 4 with the USB HDMI dongle, my video is coming out quite blurry - is there any diagnostic work I can do to check this or scope out what exactly is causing it?

Example:
[image]

mdevaev, 2020-09-09:
This is one of the problems that I don't recommend using the dongle (as described in the documentation). This is a dongle hardware problem, it can't be fixed. If the resolution in the System menu in Pi-KVM is set correctly and corresponds to the system one on your workstation, then nothing else can be done. Sorry.

closed 2020-09-10 by maxexcloo

mdevaev, 2020-09-10:
A little comparison to confirm my words.

Top: HDMI to CSI-2 Bridge (~$35 USD) / GPU comression
Middle: HDMI Game Capture (~$60 USD) / HW compression
Bottom: HDMI USB Dongle (~$20 USD) / HW compression

[image]

maxexcloo, 2020-09-10:
Great comparison 👍
I've ordered the CSI-2 bridge and a Pi Zero W for my use case :)