pikvm/pikvm #61: Network Settings within the UI

Comtech80, 2020-09-10:
Hi, It would be nice to add network setting section within the webpage to give us the ability to change pikvm setting from DHCP to a static address and gateway without going into the console.

n1nj4888, 2020-10-01:
To add to this, it would be useful to be able to configure Wifi SSIDs and passwords to be remembered in the Web UI also so that, for example pikvm built with wifi SSID A, user can go into web gui and add another SSID to be remembered so that when the pi is moved to different location, the pi connects to the new network automatically.

wnark, 2021-03-08:
> To add to this, it would be useful to be able to configure Wifi SSIDs and passwords to be remembered in the Web UI also so that, for example pikvm built with wifi SSID A, user can go into web gui and add another SSID to be remembered so that when the pi is moved to different location, the pi connects to the new network automatically. I think this essentially means a basic UI for NetworkManager

I remember that the wifi configuration of the Raspberry Pi can remember several WiFis. In theory, the web can also add several WiFi account passwords.

geerlingguy, 2022-02-04:
This would be quite helpful as the current method (https://docs.pikvm.org/wifi/) is a bit iffy sometimes, and relies on a lot of copying and pasting (and can go wrong).

vnd, 2026-09-21:
> This would be quite helpful as the current method (https://docs.pikvm.org/wifi/) is a bit iffy sometimes, and relies on a lot of copying and pasting (and can go wrong).

On a related note, apparently WiFi instructions are missing the last step to apply configuration immediately (rebooting or something like 'networkctl reload && networkctl reconfigure wlan0').

mdevaev, 2026-09-21:
> On a related note, apparently WiFi instructions are missing the last step to apply configuration immediately (rebooting or something like 'networkctl reload && networkctl reconfigure wlan0').

Fixed, thank you.