pikvm/pikvm #63: Loops for macros

tryallthethings, 2020-09-10:
Minor feature request: The ability to add loops in macros. The idea would be an option to loop the whole macro `forever until stopped`, `x times`, `for x minutes` or looping just parts of the macro.
Might be a helpful feature for automation or automated testing.

mdevaev, 2020-09-11:
Right now, the format doesn't allow for loops, and to tell you the truth, I wouldn't want to turn it into a programming language. At least at this stage. I would like to add a feature when it has a clear real case that cannot be solved otherwise.

But I can add a switch that will loop the playback of the finished macro. I believe this will partially solve problem #57.

mdevaev, 2020-09-11:
Done

closed 2020-09-11 by mdevaev

tryallthethings, 2020-09-11:
You're amazing :+1: 
The idea was to just have it here for discussion and implemented at some point in the distant future. Having a subset implemented already is even better. Keep it up!