pikvm/pikvm #69: Invalidate user sessions upon password change

tryallthethings, 2020-09-12:
Security feature: If you change the password via `kvmd-htpasswd set admin` any active user session should be invalidated / logged out.

mdevaev, 2020-09-13:
Invalidation of sessions will not result in closing connections. this is especially true for third-party applications such as ttyd. I suggest simply reflecting the current behavior in the documentation. Like, if you need to reset all sessions urgently restart kvmd and nginx.

tryallthethings, 2020-09-13:
Hardly anyone would check the documentation for a password change. Especially since the command to use is listed upon SSH login.  IMHO it would be better to ask the user if they want to restart the services after the password change if that's the only way to "kick" them out.
```
[root@pikvm ~]# kvmd-htpasswd set admin
Password:
Repeat:
Note: Users logged in with the username "admin" will stay logged in. Do you want to restart KVMD and nginx to log them out? [y/N] y
Restarting kvmd ...
Restarting nginx ...
[root@pikvm ~]# 
```

I'll raise a documentation PR later including this information.

mdevaev, 2020-09-13:
Good idea.

mdevaev, 2020-09-14:
After discussion, we decided not to change the existing behavior and just add a warning.

[image]

mdevaev, 2020-09-14:
Done

closed 2020-09-14 by mdevaev