pikvm/pikvm #73: First boot auto configuration / Enterprise deployment

tryallthethings, 2020-09-16:
Feature request:
Automatic first boot configuration such as network, hostname, override.yaml and others.

Is it possible to run a script on first boot? If so, we could add a first boot configuration script.

My proposal is a 2-step implementation:

### Step 1: Automated configuration
A script that runs on first boot that reads a configuration file from the boot partition (since that's the only one available for Windows to write on) and sets up the device configuration accordingly. The configuration may include: 
- Network configuration (IP, subnet mask, DNS, hostname, etc.)
- set override.yaml configuration parameters or just move an existing file from the boot partition
- add users and / or passwords for system / kvmd
- run system update / add packages
- replace SSL certificate (internal CA or Let's Encrypt)
- add Pi-KVM device to network monitoring system (Prometheus, Nagios, PRTG, etc.)
<hr>
- The configuration file should be an easy to parse / validate format (YAML, XML, etc.)
- should generate a log to document steps and any problems.
- Configuration file can be generated manually, via website or desktop tool

### Step 2: Enterprise deployment
Same as above but instead of placing a file on the boot partition there'll be a server on the network that provides the configuration via API. On the server you can configure a range of IP-addresses to be used by the Pi-KVM, user(s) to be added, override.yaml configuration, configuration groups, etc. 
- The server should log the outcome of the configuration process, MAC address of configured the Pi, [...]
- Server discovery could happen through broadcast, DHCP option or static hostname.

I could probably write said script and server side but I couldn't find a way to only run it once on first boot.

**Feel free to add your own ideas to this.**

ndastur, 2020-12-12:
I was going to add an issue about wifi config without needed to connect to wired eth first. Going to be covered by this, but could I make a request for a config file that at least configures wifi on first boot.

daxelrod, 2021-02-18:
I’ve implemented the beginnings of this. It is now possible to edit pikvm.txt on the boot partition to [specify the WiFi SSID and password](https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md#the-final-steps).

This is implemented in the [pikvm-bootconfig](https://github.com/pikvm/os/blob/master/pikvm-image/_pikvm-bootconfig.sh) service which can be extended to support other kinds of configuration as well.

mdevaev, 2021-11-08:
New images have `/etc/kvmd/override.d` in which you can stack override files that will be layered on top of each other. This should greatly simplify the goals of this topic.

mdevaev, 2021-11-08:
I thought about this task for a long time and came to the conclusion that each enterprise is unique like a snowflake and solutions that are suitable for one infrastructure will not suit another. So the most true and universal way would be to provide the ability to run your own executables at boot.

The new images will provide the ability to run them in alphabetical order from the `/etc/kvmd/bootconfig.d` directory. They will be executed if the file exists `/boot/pikvm.txt`.

This way we will cover absolutely all needs and will not impose a specific solution on everyone.

mdevaev, 2021-11-08:
This will also allow you to download some scripts over the network and run them. Enterprise admins who require the *Stage 2* will have no difficulty writing a few dozen lines in bash for integration into their infrastructure.

closed 2021-11-08 by mdevaev