pikvm/pikvm #74: Annoying question when returning from KVM to menu

tryallthethings, 2020-09-17:
Bug or feature?
When I'm returning from https://pikvm-1/kvm/ back to the menu (https://pikvm-1/) via the top left Pi-KVM logo / button I'm always greeted with a question:
* This page is asking you to confirm that you want to leave - data you have entered may not be saved. (Firefox)
* Changes you made may not be saved. (Chrome)

Is there a way around it or is it a necessary "feature" of the way keys are sent to the controlled computer?

mdevaev, 2020-09-17:
Yes, this is a feature to avoid accidentally closing the session.

mdevaev, 2020-09-17:
PS: This is not related to key transmit

tryallthethings, 2020-09-17:
Okay. Let's assume I sometimes know what I'm doing and don't need this feature. Can I disable it (override.yaml or otherwise)?

mdevaev, 2020-09-17:
This is not possible right now, but I will add.

tryallthethings, 2020-09-17:
Thanks. Lower than low priority :smile:

mdevaev, 2020-09-19:
Currently, the web interface does not have any mechanisms for saving settings, so I drop these parameters in meta.yaml. This may change in the future, but you can still use it.
`/etc/kvmd/meta.yaml`:
```yaml
web:
    confirm_session_exit: false
```
Will be available in the next release.

closed 2020-09-19 by mdevaev