pikvm/pikvm #75: using VNC without video, PIKVM for blind users?

a-singer, 2020-09-18:
Hello,

I'm sorry to raise this here, I'm not sure if it's an "issue", but I'm having problems using the Discord. 
	I am totally blind and don't require the video for the RPI, all I need is the keyboard input. Will not using any HDMI connection be possible? Is it required that I purchase and hook up an HDMI connection from a PC to the pi? Thanks.

mdevaev, 2020-09-18:
Hello!

Oh, that's a difficult question. In general, Pi-KVM can work if you don't connect a video device. The video window will just be empty and there will be a lot of information in the logs behind the scenes that the video device was not found. But the keyboard and other functions will work.

However, how are you going to get feedback? Pi-KVM does not support sound and Braille display. The first will be implemented, but the second is currently fundamentally impossible (due to the fact that only images go back from the server to pikvm).

I'll be happy to help you if I can.

a-singer, 2020-09-18:
Let me first thank you for the rapid and comprehensive reply. I just bought the RPI 4 piece of this puzzle after reading it. I don't mean to clutter your issues with unconnected stuff, but I didn't plan to use PIKVM for sound or braille, though I would, of course, be overjoyed to see low-latency sound functionality. Sadly, I have yet to see a VNC client which supports audio, though I have seen it with Microsoft's RDP. Anyhow, just to answer the question, I plan to use either TeamTalk
http://www.bearware.dk
or SoundWire
http://www.georgielabs.net
Both provide low latency audio from the machine, the latency is enough to type with. As for braille, I plan to use VirtualHere
http://www.virtualhere.com
though this will only work for Android, Linux, and Windows clients.
Thanks again, feel free to close this issue/question.

mdevaev, 2020-09-18:
I understood. Well, like I said, the keyboard will work. I wouldn't call your problem unconnected, it's still your case. In any case, I plan to implement audio support in the future. If you have more questions, you can ask them here or in a separate issue.