pikvm/pikvm #76: Using USB 3.0 hub rather than power splitter?

a-singer, 2020-09-18:
Hello,
Again, I'm sorry if this is for the Discord, I have difficulty using Discord. I have a USB 3.0 hub which has power pass-through, two type A USB ports, an SD card reader, and an HDMI output. If I were to plug this into an RPI4 with the PIKVM image, would it allow me to use an A to A cable to connect to a PC's USB ports as a keyboard or mouse? Alternatively, would it allow me to use an A to C cable to connect to a computer's USB port as a keyboard or mouse? What I'm after is to avoid the power splitter noted on the main page, I have purchased it but it is still shipping and looks, from the state of our mail, like it will take a while. I would try this, but don't have the RPI4 here yet. Does anyone have any experience with this? Thanks.

mdevaev, 2020-09-18:
Hi. You can insert the Pi with a USB-C to A cable into your hub, then connect the hub to your computer with an A cable. This should work if your hub does have external power.

mdevaev, 2020-09-22:
This issue seems to be resolved. If you have any problems or want to ask something else, you can re-open this issue or create a new one. I will be happy to help you.

closed 2020-09-22 by mdevaev

a-singer, 2020-09-24:
Hello,

Thanks for your work on this. I'm having no success with a hub, but may very well be doing it wrong. My hub is at

https://www.amazon.ca/ICZI-Multiport-Delivery-Charging-Chromebook/dp/B07HC3SYVK/

I plug this hub into the Pi's USB C port, and then plug power into this hub. The Pi powers on and I am able to visit the web interface without trouble. I run an A to A cable to my PC, which detects nothing. Similarly, I run an A to C cable from the hub to another PC and, again, the PC does not show that anything has been plugged in. I hit enter on "reset keyboard and mouse", and am asked whether I wish to reset the keyboard and mouse. When I answer that I do, nothing happens. I suspect that the above hub is not suitable, but would be most grateful if someone could have a look and see if it's a wiring/setting issue that is causing the problem, or if I am correct that the hub itself is the issue.
Thanks,