pikvm/pikvm #78: No BIOS/UEFI access on GA-H77-DS3H

zsmbrvr, 2020-09-21:
I get a blank/black screen when I try to enter the BIOS/UEFI or Boot option menu on GA-H77-DS3H.

The Web UI shows that the TC358743 has a resolution of 1280x720, But there's no signal.

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform v2-hdmi
  - Video capture typeCSI bridge - TC358743 
  - KVMD version:  kvmd 1.99-1, kvmd-platform-v2-hdmi-rpi4 1.99-1, kvmd-webterm 0.34-1
  - uStreamer version: ustreamer 1.25-1
  - Linux kernel: Linux pikvm 5.4.61-2-ARCH #1 SMP PREEMPT Fri Sep 4 12:46:16 UTC 2020 armv7l GNU/Linux

**Additional context**
GPU is a Sapphire RX480 Nitro+ 4GB
CPU Xeon E3-1230v2 (no iGPU)

I think the issue might be with the EDID info the computer gets from the TC358743.

mdevaev, 2020-09-21:
Try connecting the monitor and see in its settings what resolution, frequency, and other parameters are transmitted during UEFI.

mdevaev, 2020-09-21:
In addition to pikvm, do you have a connected monitor or hdmi dummy plug?  In other words, is pikvm the only video output device?

zsmbrvr, 2020-09-21:
The Pi is the only one connected.
Sadly the only monitor I've got shows 1080p as UEFI resolution, but it is clearly wrong, since Windows boot is stretched.

mdevaev, 2020-09-21:
Wait, didn't you come to discord with this problem?)

zsmbrvr, 2020-09-21:
Tried multiple EDID, boot splash is still 1280x720.

Can boot Ubuntu ISO.

zsmbrvr, 2020-09-21:
Well, after spending a few hours with @mdevaev trying to find a solution and giving up for the night, I managed to solve it.
I didn't realise that I was setting the EDID temporarily and a reboot of the PI would reset it.

I finally made it work with this EDID:

`00 FF FF FF FF FF FF 00 52 62 65 87 00 88 88 88
1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26
0F 50 54 25 40 00 01 00 01 00 01 00 01 00 01 00
01 00 01 01 01 01 D5 1B 00 50 50 00 19 40 08 20
B8 00 80 00 10 00 00 1E EC 2C 80 A0 70 38 1A 40
30 20 35 00 40 44 21 00 00 1E 00 00 00 FC 00 50
49 2D 4B 56 4D 20 56 69 64 65 6F 0A 00 00 00 FD
00 32 3D 0F 2E 0F 00 00 00 00 00 00 00 00 00 72
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

All it does is set 1280x1024 as recommended / default resolution and the BIOS accepts that.

closed 2020-09-21 by zsmbrvr

mdevaev, 2020-11-03:
Another EDID for Intel NUC:
```
00 FF FF FF FF FF FF 00 52 62 88 88 00 88 88 88 1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26 0F 50 54 25 40 00 01 00 01 00 01 00 01 00 01 00 01 00 01 01 01 01 D3 2C 80 A0 70 38 1A 40 30 20 35 00 40 44 21 00 00 1E 7E 1D 00 A0 50 00 19 40 30 20 37 00 80 00 10 00 00 1E 00 00 00 FC 00 50 49 2D 4B 56 4D 20 56 69 64 65 6F 0A 00 00 00 FD 00 32 3D 0F 2E 0F 00 00 00 00 00 00 00 00 01 C4 02 03 04 00 DE 0D 20 A0 30 58 12 20 30 20 34 00 F0 B4 00 00 00 18 E0 15 00 A0 40 00 16 30 30 20 34 00 00 00 00 00 00 18 B4 14 00 A0 50 D0 11 20 30 20 35 00 80 D8 10 00 00 18 AB 22 A0 A0 50 84 1A 30 30 20 36 00 B0 0E 11 00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 45
```

RUHavingFun, 2021-09-02:
Having a similar issue with GA-Z77X-UP4.  The first EDID helps correct the initial BIOS screen.  I get 800x600.  However, still goes black when entering the BIOS.  I have downloaded the EDID tool.  But not at all sure how to produce a good new EDID hex string.  I have tried importing the first one above and modifying and exporting to get the HEX.  But still no luck.

mdevaev, 2021-09-03:
@RUHavingFun could you try this?
```
00 FF FF FF FF FF FF 00 52 62 88 88 00 88 88 88
1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26
0F 50 54 3F CD 00 81 00 81 44 95 84 81 FC B3 84
81 84 71 44 01 01 01 1D 00 72 51 D0 1E 20 6E 28
55 00 C4 8E 21 00 00 1E 8C 0A D0 8A 20 E0 2D 10
10 3E 96 00 13 8E 21 00 00 1E 00 00 00 FC 00 54
6F 73 68 69 62 61 2D 48 32 43 0A 20 00 00 00 FD
00 3B 3D 0F 2E 0F 1E 0A 20 20 20 20 20 20 01 84
02 03 1D 03 4E 04 13 03 02 12 11 01 20 21 A2 3C
3D 3E 1F 66 03 0C 00 30 00 80 E2 00 7F 8C 0A D0
8A 20 E0 2D 10 10 3E 96 00 C4 8E 21 00 00 18 8C
0A D0 8A 20 E0 2D 10 10 3E 96 00 13 8E 21 00 00
18 97 29 A0 D0 51 84 22 30 50 98 16 00 9A 01 11
00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 6E
```

RUHavingFun, 2021-09-03:
Sadly that went straight to black as well.

mdevaev, 2021-09-03:
Try to update your BIOS

RUHavingFun, 2021-09-03:
I am at the latest "F9" from 2012/10/24.  (https://www.gigabyte.com/Motherboard/GA-Z77X-UP4-TH-rev-10/support#support-dl-bios).

I wish I understood the EDID tool better to experiment with it more.  I imported what you sent:

[image]

mdevaev, 2021-09-03:
Try this:
```
00 FF FF FF FF FF FF 00 52 62 88 88 00 88 88 88
1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26
0F 50 54 3F CD 00 01 00 01 00 01 00 01 00 01 00
01 00 01 00 01 00 8C 0A D0 8A 20 E0 2D 10 10 3E
96 00 13 8E 21 00 00 1E 00 00 00 10 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 FC 00 54
6F 73 68 69 62 61 2D 48 32 43 0A 20 00 00 00 FD
00 3B 3D 0F 2E 0F 1E 0A 20 20 20 20 20 20 01 A6
02 03 1D 03 4E 04 13 03 02 12 11 01 20 21 A2 3C
3D 3E 1F 66 03 0C 00 30 00 80 E2 00 7F 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0E
```

RUHavingFun, 2021-09-03:
That one has by far been the most unstable - screen is choppy and flashes.  I can see the splash but the BIOS is still black.

zsmbrvr, 2021-09-03:
@RUHavingFun how are you applying the EDID to the Pi?

Are you doing this:

in your favourite editor save the above as

`/root/edid.hex`
Then:
`v4l2-ctl --device=/dev/video0 --set-edid=file=/root/edid.hex --fix-edid-checksums`

DO NOT REBOOT the Pi. Just your PC.

RUHavingFun, 2021-09-03:
Yes.

RUHavingFun, 2021-09-03:
If I boot to a monitor then pull the HDMI and put in the PiKVM it is happy.

[image]

zsmbrvr, 2021-09-03:
What GPU do you have?

RUHavingFun, 2021-09-03:
I took out my video card and am using the on board GPU (Intel DSL3510L).  In the very beginning I had the same issue with the video card I had in the system too.  I took it out to eliminate it as a cause.  I have also tried the DVI port, VGA, adapters, and different cables.

mdevaev, 2021-09-03:
So, 1280x1024. Try this:
```
00FFFFFFFFFFFF005262888800888888
1C150103800000780AEE91A3544C9926
0F50542FCF0001000100010001000100
0100010101018C2300A050001E403020
370080001000001E000000FC0050492D
4B564D20566964656F0A000000FD0032
3D0F2E0F000000000000000000000010
0000000000000000000000000000016B
02030400DE0D20A03058122030203400
F0B400000018E01500A0400016303020
3400000000000018B41400A050D01120
3020350080D810000018AB22A0A05084
1A3030203600B00E1100001800000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000045
```

RUHavingFun, 2021-09-03:
That looks like the one you suggested above for the NUC.  I will try it now.  I noticed it is different in spacing.  Are the spaces between HEX sets important or does it read them fine?  I assume maybe it will, as I noted it seems fine with block format or one long string as well.

RUHavingFun, 2021-09-03:
Yahoo!  After comparison I see they are in-fact not the same.  And this one worked.  I am about to pull into the editor to see if I can tell what you did.  Thank you very much!  Hopefully this can help others as we learn what was the issue.

mdevaev, 2021-09-03:
I removed the timing for 1080p. Perhaps the BIOS is not behaving quite correctly and is trying to do something unsupported. I think that I need to make an EDID switch in the interface. We can experiment more.

PS: Spaces are not important

RUHavingFun, 2021-09-03:
Thank you!  (I figured they were not)  I am willing to help test if I can be of assistance.

mdevaev, 2021-10-01:
We have planned this and will make an EDID switch menu, but I don't know when yet.

Modac, 2022-04-20:
> So, 1280x1024. Try this:
>
> ```
> 00FFFFFFFFFFFF005262888800888888
> 1C150103800000780AEE91A3544C9926
> 0F50542FCF0001000100010001000100
> 0100010101018C2300A050001E403020
> 370080001000001E000000FC0050492D [...]

Just wanted to let you know that no edid worked for me but this one.
When I entered the BIOS it just showed a black screen, everything else worked.
My motherboard is Gigabyte GA-Z68XP-UD3 (rev. 1.3) with BIOS version `U1J (UEFI BIOS)`.

priddyma, 2022-06-11:
Just posting for anyone who may come across this / Google:
The 1280x1024 fixed my ASUS Z390-E Gaming giving no signal. I would receive this error:
`[2022-06-11 06:52:44 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- => -- ERROR [80602.341    stream] -- Persistent device timeout (unplugged)`

ahmaddxb, 2022-06-13:
how do you edit with AW EDID Editor? I want to use below but there is no easy way I can see to input it.

00FFFFFFFFFFFF005262888800888888
1C150103800000780AEE91A3544C9926
0F505425400001000100010001000100
010001010101D32C80A070381A403020
350040442100001E7E1D00A050001940
3020370080001000001E000000FC0050
492D4B564D20566964656F0A000000FD
00323D0F2E0F000000000000000001C4
02030400DE0D20A03058122030203400
F0B400000018E01500A0400016303020
3400000000000018B41400A050D01120
3020350080D810000018AB22A0A05084
1A3030203600B00E1100001800000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000045

priddyma, 2022-06-13:
https://github.com/pikvm/pikvm/issues/78#issuecomment-912778447

asheroto, 2022-07-22:
> Well, after spending a few hours with @mdevaev trying to find a solution and giving up for the night, I managed to solve it. I didn't realise that I was setting the EDID temporarily and a reboot of the PI would reset it.
> 
> I finally made it work with this EDID:
> 
> `00 FF FF FF FF FF FF 00 52 62 88 88 00 88 88 88 1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26 0F 50 54 25 40 00 01 00 01 00 01 00 01 00 01 00 01 00 01 01 01 01 D5 1B 00 50 50 00 19 40 08 20 B8 00 80 00 10 00 00 1E EC 2C 80 A0 70 38 1A 40 30 20 35 00 40 44 21 00 00 1E 00 00 00 FC 00 50 49 2D 4B 56 4D 20 56 69 64 65 6F 0A 00 00 00 FD 00 32 3D 0F 2E 0F 00 00 00 00 00 00 00 00 01 4D 02 03 04 00 DE 0D 20 A0 30 58 12 20 30 20 34 00 F0 B4 00 00 00 18 E0 15 00 A0 40 00 16 30 30 20 34 00 00 00 00 00 00 18 B4 14 00 A0 50 D0 11 20 30 20 35 00 80 D8 10 00 00 18 AB 22 A0 A0 50 84 1A 30 30 20 36 00 B0 0E 11 00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 45`
> 
> All it does is set 1280x1024 as recommended / default resolution and the BIOS accepts it.
> 
> An alternative is this: `00 FF FF FF FF FF FF 00 52 62 88 88 00 88 88 88 1C 15 01 03 80 00 00 78 0A EE 91 A3 54 4C 99 26 0F 50 54 25 40 00 01 00 01 00 01 00 01 00 01 00 01 00 01 01 01 01 D3 2C 80 A0 70 38 1A 40 30 20 35 00 40 44 21 00 00 1E 7E 1D 00 A0 50 00 19 40 30 20 37 00 80 00 10 00 00 1E 00 00 00 FC 00 50 49 2D 4B 56 4D 20 56 69 64 65 6F 0A 00 00 00 FD 00 32 3D 0F 2E 0F 00 00 00 00 00 00 00 00 01 C4 02 03 04 00 DE 0D 20 A0 30 58 12 20 30 20 34 00 F0 B4 00 00 00 18 E0 15 00 A0 40 00 16 30 30 20 34 00 00 00 00 00 00 18 B4 14 00 A0 50 D0 11 20 30 20 35 00 80 D8 10 00 00 18 AB 22 A0 A0 50 84 1A 30 30 20 36 00 B0 0E 11 00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 45` 1920x1080 as recommended / default resolution and the BIOS accepts it.
> 
> To test:
> 
> in your favourite editor save the above as
> 
> `/root/edid.hex` Then: `v4l2-ctl --device=/dev/video0 --set-edid=file=/root/edid.hex --fix-edid-checksums`
> 
> DO NOT REBOOT the Pi. Just your PC. Check if the BIOS boot splash is displayed at 1280x1024 OR 1920x1080.

I am using a PiKVM to prep some Dell OptiPlex computers. I found that it worked fine except it would not load the BIOS/UEFI or the boot menu. Odd thing is - if I plug in a regular monitor and load BIOS/UEFI, and _then_ switch the cables, it's fine, but would not work directly.

Using this hex config fixed the issue, so thank you!

asheroto, 2022-07-22:
Okay, that being said, that command does make the BIOS/UEFI show up properly.

How do I set it permanent using `kvmd-edidconf` ? I know I can copy it in place of `tc358743-edid.hex` but didn't want to do that if it wasn't _quite_ a perfect hex file?

I have read [this page](https://docs.pikvm.org/edid/#editing-edid) but wasn't sure the best way to handle this?

mdevaev, 2022-07-22:
You can overwrite `tc358743-edid.hex`. The default file can be found in `/usr/share/kvmd/configs.default/kvmd/tc358743-edid.hex`

asheroto, 2022-07-22:
Okay, thanks. Does the formatting matter that it is different (like in the screenshot)?

mdevaev, 2022-07-22:
No

mdevaev, 2022-07-22:
:ok_hand:

TheLinuxGuy, 2022-09-01:
> So, 1280x1024. Try this:
>
> ```
> 00FFFFFFFFFFFF005262888800888888
> 1C150103800000780AEE91A3544C9926
> 0F50542FCF0001000100010001000100
> 0100010101018C2300A050001E403020
> 370080001000001E000000FC0050492D [...]

This one worked for me to fix BIOS screen on Intel 12th gen MSI PRO-A DDR4 B660 motherboard. 

It would be a nice addition to wiki page discussing this EDID problem, and showing an example of original EDID in hex could help restore settings back to original.

priddyma, 2022-09-01:
It is explained in a section. Also has reference to the stock file location
on the system.

stephendt, 2023-01-04:
I also just experienced this issue, very peculiar indeed. The EDID posted by [Modac](https://github.com/Modac) did the trick. An EDID switch would be a very appreciated feature.

Majestic7979, 2024-01-17:
> Well, after spending a few hours with @mdevaev trying to find a solution and giving up for the night, I managed to solve it. I didn't realise that I was setting the EDID temporarily and a reboot of [...]

Thank you. Here's the file in case someone needs it (can be opened in notepad from Windows or cat on linux to verify the contents):
[myedid.zip](https://github.com/pikvm/pikvm/files/13968879/myedid.zip)

asheroto, 2024-01-17:
Cool! Thanks.

Should we still use the EDID files with...

```bash
v4l2-ctl --device=/dev/video0 --set-edid=file=/root/edid.hex --fix-edid-checksums
```

or use the new `kvmd-edidconf` tool?

Does it depend on the use case?

Just to confirm, the `kvmd-edidconf` tool keeps it set permanently? How can it be undone if so? 😊

Thanks so much!