pikvm/pikvm #79: wiring RPI 4 with power splitter?

a-singer, 2020-09-24:
Hello,

	I'm sorry to ask this, I am totally blind and it is probably for this reason that I'm having the issue. That is, I'm sure the diagram on the main site solves this problem. I recently ordered the device at

https://www.tindie.com/products/8086net/usbpwr-splitter/

	The product arrives and seems to be a circuit board with four ports. First, a USB A female port, then a round barrel type connector, and finally two micro B female ports. I suspect that what I need to do is to connect the USB A female port to the Raspberry Pi 4's USB c port, and to connect the two Micro B ports to the computer the KVM is supposed to control. However, I am unsure as to how to bring power to the Pi. I thought there would be a female USB C port on the circuit board I ordered but there is no such port, there is only the round barrel female port to plug into. Should I use that port to bring in power and, if so, what power supply is to be used? Also, this is not mentioned in the required hardware list, so I may well be missing something. How do I connect the Pi to power using this board?

Thanks,

mdevaev, 2020-09-24:
Turn the board with the micro connectors away from you. The side with the connectors should be on top.

USB A: connect to Raspberry Pi.
Left Micro USB port: connect to the power.
Right Micro USB port: connect to the server.
Barrel port: just ignore.

mdevaev, 2020-09-24:
Edited.

a-singer, 2020-09-24:
Thank you, this description is excellent and both the VNC and web interface are working. The server detects the HID keyboard and typing on the VNC client is reflected on the computer controlled by KVM. Thanks again.

closed 2020-09-24 by a-singer