pikvm/pikvm #80: remapping keys through VNC

a-singer, 2020-09-24:
Hello,

	If I may ask, is there a way to remap keys on pikvm? Certain applications use the capslock as a modifier key. Unfortunately, the VNC clients I have found when they see capslock down, simply send a shifted version of whatever they see. That is, if I hit capslock+a on the keyboard, my VNC client sends shift+a, not capslock+a. Would it be possible to remap, for example, the right alt key to capslock, so that whenever pikvm sees rightalt+a, it will send capslock+a to the USB port? Thanks.

mdevaev, 2020-09-24:
Unfortunately this cannot be done. Any remapping you need to do on the host side.

mdevaev, 2020-11-05:
Is it still relevant?

mdevaev, 2020-11-12:
Sup?

a-singer, 2020-11-13:
Sorry, didn't check the email for this account for a while. As the starter of the issue, this isn't relevant to me any longer. Obviously, others may find it so, but I find that because PiKVM looks like a USB keyboard, I can just modify a key through the Windows registry and have done with it. While it would be nice to do this on the KVM itself, both because it would make it easier to move from machine to machine and it would make it easier to have the PiKVM plugged into a multi keyboard switch, it's definitely just a nice to have for me at this point. Thanks for looking into this.

arwarw, 2021-04-28:
Slightly different scenario for me: I'd like to remap CapsLock to Ctrl. However, neither the webclient nor the vnc client seem to do that, even with a reefined keymap (vnc.keymap & kvmd.hid.keymap: /usr/share/kvmd/keymaps/en-nocaps which is the same as en-us, just the Caps_Lock 0x3a replaced by Control_L 0x3a). Does the webclient have the same limitation for remapping Capslock as the VNC clients do?

Sorry if I mistakenly hijacked an unrelated bug, but it seems very similar to me.

mdevaev, 2021-04-28:
Web UI doesn't use this keyfiles so you can't remap keys in this way.

arwarw, 2021-05-04:
Is it currently possible to currently remap CapsLock to Control in either the Web Client or VNC? If so, how?

mdevaev, 2023-10-24:
@arwarw Hey, I lost your request for this task and only now saw it when I was going through the list. I hope the "better late than never" rule will work this time, because in the latest version of KVMD I added an option to remap Ctrl and Caps.

closed 2023-10-24 by mdevaev

arwarw, 2023-11-06:
Uh, cool!

Definitely still relevant, will test asap. Thanks alot @mdevaev !

arwarw, 2023-11-06:
Works great! :smile: :partying_face: 

ffr: https://github.com/pikvm/kvmd/commit/3c611121a8e077e55478b9c55b51221f25207a9d

mdevaev, 2023-11-06:
:ok_hand: