pikvm/pikvm #81: Windows key not working when connecting to VNC server

a-singer, 2020-09-24:
Hello,

	When connecting to pikvm through VNC, I am unable to use the windows key from the VNC client. It is as if the windows key is not pressed at all, either as a modifier or on its own. When I connect from the same client to another VNC server, I am able to hit windows and it behaves normally. I made no changes to the keyboard layout, only following instructions 1, 4, and 5 from the vnc.md instructions. If I may ask, what is the best method to enable the windows key? Thanks.

mdevaev, 2020-09-24:
Which client do you use?

a-singer, 2020-09-24:
Thanks for the quick response. I use Jump Desktop for iOS
https://jumpdesktop.com/
The program does send the windows key to another Windows VNC server I work with and does so without trouble. However, I may say I don't have another Linux VNC server to experiment with. Thanks again.

mdevaev, 2020-10-03:
Confirmed.

mdevaev, 2020-10-03:
Fixed. Will be released soon.

closed 2020-10-03 by mdevaev

a-singer, 2020-10-03:
I just wanted to confirm that this is fixed. I replaced the mappings file with the new one and hitting the windows key on the VNC client sends that to the server. Thanks very much.

mdevaev, 2020-10-04:
Released. Update using:
```
rw
pacman -Syu
reboot
```