pikvm/pikvm #83: Particular use case for blind users or, how to save $1030 USD while getting better results through pikvm

a-singer, 2020-09-27:
Hello,

	This is not really an issue, more a message of thanks, but may be  something the developers ought to know. I understand the point of pikvm is to support companies with large fleets of machines. However, I have two machines here and am running them off a cheap USB KVM switch into which my pi is plugged. I am totally blind and my screen reader, Jaws, supports a very limited range of remote control options. To use remote access at all, one must pay $1000 USD for a perpetual license, then pay $200 for a remote desktop license, and then be permitted to connect only from Windows or through Citrix. You cannot properly connect from iOS or Android because Jaws will not recognise any command sent through VNC, it seems to hook the keyboard at a lower level and insist on the keyboard being connected via USB HID or PS/2. If you attempt to use RDP without the remote access authorization, whether from Windows or any other client OS, it will fail. If you attempt to use VNC from anywhere, it will fail to accept commands. Basically, you can't use a mobile device for real remote access at all and can only use a Windows machine after having paid $1200. You can use Virtual here to connect a virtual USB keyboard but not on iOS. Even if mobile remote was workable, the cost puts it out of reach for most. Having said that, pikvm solves the problem. One can use any VNC client from any client OS and it looks like a normal keyboard to Jaws. Everything just works. The only change I had to make was to remap the right alt key to serve as the capslock or Jaws key. The sound can be brought over using multiple methods, I use a conferencing system which uses Opus. The total cost of this system, laying aside time, is approximately $170 ($90 for a Jaws home license which I would need anyway and $80 for the Pi/card/USB splitter/shipping). This is a huge step forward in a way which is somewhat unusual, so I thought I had better raise it both as an odd thing for the developers and for people who might want to do this using screen readers in the future. I appreciate the work that has been done on this, it works very well and smoothly.

mdevaev, 2020-09-27:
Thank you for your words! I'm glad I helped you solve your problems. By the way, I didn't forget about your other issues. The fact is that I'm currently the only Pi-KVM developer and I'm busy with a critical bug related to the Linux kernel and keyboard. As soon as I get this sorted out, I'll help you with the rest of your cases.

a-singer, 2020-09-27:
Hi, 

First, I am absolutely astounded that one person could create such a frictionless project. I appreciate it.
Secondly, the issues I'm having are important, in my undoubtedly biased judgment, but are not terribly urgent.  Both can be worked around and I have done that with both, but at the same time, the workarounds themselves aren't perfect so I would be grateful if you could have a look at solutions for them when you have a minute. Thanks again.

closed 2020-09-27 by a-singer

tryallthethings, 2020-10-01:
mdevaev deserves all the praise you gave him. It really is an outstanding project!
I honestly don't fully understand your particular use case as I've only had to support a blind user once very long ago and never got another chance to dive deeper into the topic. Is the setup you're using maybe useful for others and does it require anything besides the documented steps to set up? If so, would you mind writing it down in some form, so we can add it to our documentation?

a-singer, 2020-10-01:
Hello,

The short answers to your questions are maybe and yes :) Seriously, though, I don't know if this would benefit many other people, simply because there aren't that many blind people who use screen readers and are, like me, totally blind. That is, I have no use whatever for video, whether magnified or not. However, I do intend to write up a guide on how to do this, it shouldn't take long and even the chance that it will help one or two people is worth the short amount of time it might take. If I post it to this issue once it's done, will you be so good as to have a look and, if you think it would be useful, put it into the documentation?
Thanks.

tryallthethings, 2020-10-02:
Yes, you can just post it here and I'll take it over to the documentation. I was about to ask for screenshots but that's not going to help anyone, is it? :smile: 
Thank you in advance!

mdevaev, 2020-10-05:
I'm definitely in favor of adding this description to the documentation. If we have the opportunity to help people with unusual use-cases - why not do it. There will be more than one happy user in the world.