pikvm/pikvm #84: alt+ctrl+letter does not work when connecting to VNC

a-singer, 2020-09-27:
Hello,

	I don't seem to be able to use alt+ctrl+key when connecting to the VNC server. It is as if nothing happens. Ctrl+shift+key works, as does alt+shift+key, but not ctrl+alt+key. I can use sticky keys to do this, press alt, then press ctrl, then press a key, but can't use the combination as usual. I have made no changes to the keymap, following only instructions 1, 4, and 5 on the vnc.md page. I am using the Jump Desktop VNC client on iOS, and that client does send alt+ctrl+key properly when connected to my Windows VNC server. Might changing the keymap be helpful? Thanks.

mdevaev, 2020-09-27:
Perhaps the reason is the absence of the AltGr binding. This is fixed in git, but I'll look further into what's wrong.

a-singer, 2020-09-27:
I have looked through the recent commits but have little experience on how to best handle these things, so must have missed the fix. If it would be useful to update to the new component and test, I am happy to do that. Thanks.

mdevaev, 2020-09-30:
I can't reproduce this on Tiger VNC. And right now I don't have an iOS device where I can put your client. However, I noticed that some keyboard shortcuts, such as Ctrl+Alt+X, are not really passed by the client. On the other hand, Ctrl+Alt+Space and others work. I believe that the client's own hotkeys are not transmitted.

mdevaev, 2020-09-30:
Try other possible hotkeys. Let's find out what's wrong.

a-singer, 2020-09-30:
I'm sorry, the problem is on my end, the lack of alt+control+keystrokes is only there when VoiceOver, the iOS screen reader, is on. When I turn it off, I can hit alt+ctrl+keystroke and get it passed through to the server. Again, sorry for disturbing you with it, this should be going to Apple accessibility, I'll report it to them. I should say that the windows key issue, issue  #81  is not effected by whether the screen reader is on or off. Thanks for looking into this.

mdevaev, 2020-09-30:
Okay, I'll deal with the next problem.

closed 2020-09-30 by mdevaev