pikvm/pikvm #86: 0W version, speed and Ethernet

a-singer, 2020-10-03:
Hello,

	Pikvm has become essential to my computer use, but I would like to use the pi which is currently running it for another project. I am thinking of switching to a 0W for my keyboard only KVM. I have two questions if I may ask them. First, it says on the main page that the 0W is slower. Does this only apply to the video or will it also make the keyboard work more slowly? I cannot see a significant delay in the pi 4's keyboard relay, it seems to work very quickly, as if I am in front of the machine. Is this still the case with the 0W?
Secondly, there seem to be ways to connect Ethernet to a 0W. I realize the main page says no Ethernet, but will the 0W version of Pikvm work with these Ethernet adapters for the 0W? I would like to run the unit wired.
Thanks,

mdevaev, 2020-10-05:
You can use Zero without video, and it will be fast enough. The main problems are really because of the video.

In addition, you will be able to use the Ethernet SPI. This is a small module that connects to GPIO SPI and supports about 10 megabits.

I found an article for you that has a lot of text explanations: https://www.instructables.com/Super-Cheap-Ethernet-for-the-Raspberry-Pi/

mdevaev, 2020-10-17:
Well, I probably answered your question. Feel free to create a new if needed.

closed 2020-10-17 by mdevaev