pikvm/pikvm #90: WiFi Hotspot

Testato, 2020-10-05:
my use case for PiKvm is have a little tool in my service bag, and attach it to a server headless only momentarily
so should be useful have an Hotspot option installed on the Rpi.
In this way one can connect directly his laptop to the rpi Wifi.
In the UI should be present a switch for enable/disable Hotspot

hellsounet, 2021-02-16:
same use case for me, I managed to have it working with hostapd and dnsmasq 
pretty handy

bbeaudoin, 2021-02-23:
Another user proposed AP mode for initial wireless configuration giving this yet another feature. An example of an initial wireless config service was referenced at [sabhiram/raspberry-wifi-conf](https://github.com/sabhiram/raspberry-wifi-conf), it's also similar to how a GoPro and some other connected wireless devices are configured.

This might be combined with the default functionality of shorting pin 5 (GPIO 3) to ground for sleep/wake functionality where a short button press can toggle between host AP and normal wireless mode, a long press can shutdown the Pi. While this is normally handled with`dtoverlay=gpio-shutdown` which initiates a shutdown via systemd version 225 and later, without the overlay, the signal can be handled by an event listener.

alexander-rieder, 2021-12-10:
Our PI-KVM v3 HAT arrived today and I am really happy with it. However I also would like to use the PI-KVM in my service bag as a replacement for an external monitor when at a customers remote site. 

@mdevaev can I somehow help to implement this feature?

mdevaev, 2021-12-10:
Sup. Contact us in Discord #dev chat

saftl, 2022-01-17:
same need here for my service bag. A small pi zero only instead of keyboard, mouse and monitor would be great.

srepac, 2022-03-14:
I wrote a script to create a hotspot interface ap0 running on top of wlan0 interface of pikvm.  You don't need to have wlan0 connected to network if you're on eth0, but if your pikvm is only using wifi (i.e. pi zero (2) w), then it's advisable that you connect to SSID.

https://kvmnerds.com/PiKVM/hotspot

If you want the hotspot to always be on at boot, then run the script as

```hotspot -f```

I've fully tested this to work on zero w, zero 2w, pi4 and pi400.

SolubleSpork, 2024-01-08:
> I wrote a script to create a hotspot interface ap0 running on top of wlan0 interface of pikvm. You don't need to have wlan0 connected to network if you're on eth0, but if your pikvm is only using wifi (i.e. pi zero (2) w), then it's advisable that you connect to SSID.
> 
> https://kvmnerds.com/PiKVM/hotspot
> 
> If you want the hotspot to always be on at boot, then run the script with -f option
> 
> `hotspot -f`
> 
> I've fully tested this to work on zero w, zero 2w, pi4 and pi400.

I was using this for a long time - much appreciated.  But now, I have upgraded my home network a lot and want to reconfigure my pikvm to act more like an airconsole.  I haven't had much success because I'm really not familiar with arch, but do you think theres an easy way to use hostapd and a tool like bridge-utils to make the hotspot use the upstream DHCP from the eth0 interface?  Your script creates its own DHCP server and makes an out of bound network, but in my case, I would love to have the wireless clients be in the same net as my wired connection, all handled by my upstream router/FW.

crjmichiels, 2024-02-28:
> > https://kvmnerds.com/PiKVM/hotspot
> 
> It looks like the link is expired @srepac

Grabbed this from the Wayback Machine

```
#!/bin/bash
# Created by srepac@kvmnerds.com
#
# Filename:  hotspot
VER=1.6
## CHANGELOG:
# 1.0   03/02/22    Created
# 1.1   03/03/22    Added ability to always run services at boot (also otgnet network DHCP entry for dnsmasq)
# 1.2   03/04/22    Add ap0 interface on top of wlan0 for use as hotspot AP
# 1.3   03/11/22    Create script/service so that AP starts up on boot and make it easy to change hotspot network
# 1.4   03/12/22    Updated SSID to be hostname-AP; DNSport pick between 53 and 5553; refactoring
# 1.5   03/14/22    Use the current nameservers for use with dnsmasq
# 1.6   03/22/22    Consolidated to one script for usage on both Arch and Raspbian
#
# This script was written to allow PiKVM to run its wifi as hotspot AP akin to how GoPro is first configured
#
#   SSID:         $(hostname)-AP
#   Passphrase:   pikvmisawesome
#
#   Hotspot network IP 10.5.4.1/24   DHCP range 10.5.4.10 - 10.5.4.250
###
# Change SSID and PASSPHRASE here
SSID="$(hostname)-AP"
PASSPHRASE='pikvmisawesome'
# Replace the first 3 octets of hotspot network here (change it to whatever you want)  Default: NETWORK="10.5.4"
NETWORK="10.5.4"
###
: '
Before running this script, you should connect your wifi to SSID first, and then run this script creating ap0
... that acts like a wifi hotspot that other systems can connect to.

In addition, if the pi eth0/wlan0 is connected to internet, the systems connected to the wifi
... hotspot will also have internet access.
' ### end of comments ###

### unblock wifi in the very beginning just in case
rfkill unblock wifi

if [[ "$1" == "-h" || "$1" == "--help" ]]; then
  echo "usage:  $0 [-f] where -f forces hotspot to run at boot; default is to run on this session only"
  exit 1
fi

if [ -e /usr/local/bin/rw ]; then rw; fi
set -x

### added 03/14/22 - create list of DNS servers from /etc/resolv.conf
if [ ! -e /etc/kvmd/custom.dns ]; then
  NAMESERVERS=$( for i in `grep ^nameserver /etc/resolv.conf | awk '{print $2}' | sort -r -u`; do echo -n "$i,"; done | sed 's/,$//' )
else
  # comma separated list of DNS servers to use
  NAMESERVERS=$( egrep -v '^#' /etc/kvmd/custom.dns )
fi

### Added on 03/22/22 to allow usage on both Arch and Debian (Raspbian) ###
case $( grep ^NAME= /etc/os-release | cut -d'"' -f 2 | cut -d' ' -f1 ) in
  "Arch")
    # Install required packages if not already installed
    if [ $( pacman -Q | grep -wc hostapd ) -ne 1 ]; then
      # first, update db's
      pacman -Syy
      # then, install hostapd and dnsmasq packages
      pacman --noconfirm -S hostapd dnsmasq
[... 166 lines cut ...]

  # disable hotspot.service
  systemctl disable --now hotspot.service

fi

# Restat hostapd and dnsmasq so the config changes take effect
systemctl restart hostapd dnsmasq

### Added on 03/03/22 -- just in case kvmd-otgnet-dnsmasq was already running
# fix it by adding entry for usb0 DHCP
if [ ! -e /usr/bin/otgnet.sh ]; then
  wget -O /usr/bin/otgnet.sh  https://kvmnerds.com/PiKVM/otgnet.sh 2> /dev/null
  chmod +x /usr/bin/otgnet.sh
fi
/usr/bin/otgnet.sh

sleep 3
ip -br a | grep ap0
iwconfig ap0
systemctl status hostapd dnsmasq

if [ -e /usr/local/bin/ro ]; then ro; fi

```

wlatic, 2024-10-19:
I've slightly changed this to avoid conflicts with ipv6 and also limit the DHCP (dnsmasq) to ap0, so that if you connect to wifi AND have an AP for emergencies it still works

```
#!/bin/bash
# Created by srepac@kvmnerds.com
#
# Filename:  hotspot
VER=1.6
## CHANGELOG:
# 1.0   03/02/22    Created
# 1.1   03/03/22    Added ability to always run services at boot (also otgnet network DHCP entry for dnsmasq)
# 1.2   03/04/22    Add ap0 interface on top of wlan0 for use as hotspot AP
# 1.3   03/11/22    Create script/service so that AP starts up on boot and make it easy to change hotspot network
# 1.4   03/12/22    Updated SSID to be hostname-AP; DNSport pick between 53 and 5553; refactoring
# 1.5   03/14/22    Use the current nameservers for use with dnsmasq
# 1.6   03/22/22    Consolidated to one script for usage on both Arch and Raspbian
#
# This script was written to allow PiKVM to run its wifi as hotspot AP akin to how GoPro is first configured
#
#   SSID:         $(hostname)-AP
#   Passphrase:   pikvmisawesome
#
#   Hotspot network IP 10.5.4.1/24   DHCP range 10.5.4.10 - 10.5.4.250
###
# Change SSID and PASSPHRASE here
SSID="$(hostname)-AP"
PASSPHRASE='pikvmisawesome'
# Replace the first 3 octets of hotspot network here (change it to whatever you want)  Default: NETWORK="10.5.4"
NETWORK="10.5.4"
###
: '
Before running this script, you should connect your wifi to SSID first, and then run this script creating ap0
... that acts like a wifi hotspot that other systems can connect to.

In addition, if the pi eth0/wlan0 is connected to internet, the systems connected to the wifi
... hotspot will also have internet access.
' ### end of comments ###

### unblock wifi in the very beginning just in case
rfkill unblock wifi

if [[ "$1" == "-h" || "$1" == "--help" ]]; then
  echo "usage:  $0 [-f] where -f forces hotspot to run at boot; default is to run on this session only"
  exit 1
fi

if [ -e /usr/local/bin/rw ]; then rw; fi
set -x

### added 03/14/22 - create list of DNS servers from /etc/resolv.conf
if [ ! -e /etc/kvmd/custom.dns ]; then
  #NAMESERVERS=$( for i in `grep ^nameserver /etc/resolv.conf | awk '{print $2}' | sort -r -u`; do echo -n "$i,"; done | sed 's/,$//' )
  NAMESERVERS=$( for i in `grep ^nameserver /etc/resolv.conf | awk '{print $2}' | grep -v : | sort -r -u`; do echo -n "$i,"; done | sed 's/,$//' )

else
  # comma separated list of DNS servers to use
  NAMESERVERS=$( egrep -v '^#' /etc/kvmd/custom.dns )
fi

### Added on 03/22/22 to allow usage on both Arch and Debian (Raspbian) ###
case $( grep ^NAME= /etc/os-release | cut -d'"' -f 2 | cut -d' ' -f1 ) in
  "Arch")
    # Install required packages if not already installed
    if [ $( pacman -Q | grep -wc hostapd ) -ne 1 ]; then
      # first, update db's
      pacman -Syy
      # then, install hostapd and dnsmasq packages
      pacman --noconfirm -S hostapd dnsmasq
    fi
    ;;
[... 164 lines cut ...]
else

  # disable hotspot.service
  systemctl disable --now hotspot.service

fi

# Restat hostapd and dnsmasq so the config changes take effect
systemctl restart hostapd dnsmasq

### Added on 03/03/22 -- just in case kvmd-otgnet-dnsmasq was already running
# fix it by adding entry for usb0 DHCP
if [ ! -e /usr/bin/otgnet.sh ]; then
  wget -O /usr/bin/otgnet.sh  https://kvmnerds.com/PiKVM/otgnet.sh 2> /dev/null
  chmod +x /usr/bin/otgnet.sh
fi
/usr/bin/otgnet.sh

sleep 3
ip -br a | grep ap0
iwconfig ap0
systemctl status hostapd dnsmasq

if [ -e /usr/local/bin/ro ]; then ro; fi
```

RazielAce, 2024-12-05:
I can't get the 'script' installed, and I need a step-by-step guide. I have already followed the steps in Discord, but the GPIO didn't even appear.

Nikotine1, 2025-11-28:
I don't believe this still works.
The filesystem is read-only.

There seems to be a way to temporarily mount a writeable drive: https://docs.pikvm.org/pst/
I managed to create the script using:
`[root@pikvm data]# kvmd-pstrun nano /var/lib/kvmd/pst/data/hotspot`
`[root@pikvm data]# kvmd-pstrun chmod +x /var/lib/kvmd/pst/data/hotspot`

But running it ends in a permission denied error:
```
[root@pikvm data]# kvmd-pstrun -- /var/lib/kvmd/pst/data/hotspot --f
--    INFO -- Opening PST session ...
--    INFO -- PST write is allowed: /var/lib/kvmd/pst
--    INFO -- Running the process ...
--   ERROR -- Unhandled exception
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/kvmd/apps/pstrun/__init__.py", line 91, in _run_cmd_ws
    proc = await _run_process(cmd, event["data"]["path"])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/kvmd/apps/pstrun/__init__.py", line 57, in _run_process
    return (await asyncio.create_subprocess_exec(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    ))
    ^
  File "/usr/lib/python3.13/asyncio/subprocess.py", line 224, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
        stderr=stderr, **kwds)
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/asyncio/base_events.py", line 1794, in subprocess_exec
    transport = await self._make_subprocess_transport(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        protocol, popen_args, False, stdin, stdout, stderr,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        bufsize, **kwargs)
        ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/asyncio/unix_events.py", line 213, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
                                    stdin, stdout, stderr, bufsize,
                                    waiter=waiter, extra=extra,
                                    **kwargs)
  File "/usr/lib/python3.13/asyncio/base_subprocess.py", line 39, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                stderr=stderr, bufsize=bufsize, **kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/asyncio/unix_events.py", line 855, in _start
    self._proc = subprocess.Popen(
                 ~~~~~~~~~~~~~~~~^
        args, shell=shell, stdin=stdin, stdout=stdout, stderr=stderr,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        universal_newlines=False, bufsize=bufsize, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/subprocess.py", line 1038, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                        pass_fds, cwd, env,
                        ^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
                        gid, gids, uid, umask,
                        ^^^^^^^^^^^^^^^^^^^^^^
                        start_new_session, process_group)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/subprocess.py", line 1974, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/var/lib/kvmd/pst/data/hotspot
```